RIOT-OS/RIOT #4880: stm32: GPIO driver implementation uses magic numbers

gebart, 2016-02-23:
Example: https://github.com/RIOT-OS/RIOT/blob/df996044e2286ec0f5d24ada065520f98d03c2c5/cpu/stm32f0/periph/gpio.c#L119

These can lead to hard to find bugs and should be changed to preprocessor constants.

PeterKietzmann, 2016-02-29:
That looks like a task for @A-Paul :-) ???

A-Paul, 2016-03-10:
@PeterKietzmann, do you want me to be assigned to this issue?

PeterKietzmann, 2016-03-16:
@A-Paul no matter about the assignment. IIRC you once wanted to start an offensive to eliminate all magic numbers :-)

A-Paul, 2016-03-16:
@PeterKietzmann, in my opinion the use of numeric literals leads to some problems and should be *avoided*. Although I'm not feeling to be in charge of cleaning them up... ;)

A-Paul, 2016-03-16:
This issue has been announced in #4862 as a reminder for @haukepetersen. Are you going to address this?

DipSwitch, 2016-03-16:
I'm gonna regret this... But I think I know fix the magic numbers using the STM32 header files... Only for STM ofcourse =)

haukepetersen, 2016-03-17:
I am not going to address this anytime soon, as I personally don't have that strong opinions against magic numbers at certain places...

aabadie, 2019-09-10:
This is still an issue and could be fixed with new CMSIS. I think all STM32 cpu uses the new CMSIS versions now.

closed 2020-04-13 by stale[bot]