RIOT-OS/RIOT #4881: Is -fno-builtin required?

gebart, 2016-02-23:
Currently we build cortexm_common, msba2, arduino-mega2560 and x86-multiboot-common with 
`CFLAGS += -fno-builtin`
which eliminates the compiler builtins for many common C library functions such as memcpy, memset, printf etc. Without -fno-builtin the compiler may optimize calls to standard library functions into more efficient calls, for example, `printf("hello\n")` (with no format arguments) could be optimized into `puts("hello")`.

https://gcc.gnu.org/onlinedocs/gcc/Other-Builtins.html contains a list of some of the builtins.

I don't know the reason why the flag was added in the first place, possibly it may have interfered with some own implementation of C library functions.

I tried building without the flag on mulle and at first glance it seems to work. It even reduced the size of the gnrc_networking binary by about 2 kB!

See also: http://sjhan5.blogspot.se/2015/09/when-people-use-compile-option-fno.html

Kijewski, 2016-02-23:
For x86-multiboot-common GCC assumed that it could use some internal glibc functions when compiled without `-fno-builtin` that were not provided by Newlib (but it could be that my configuration was just wrong / less than optimal). Out of my head I don't recall which function that were.

I don't know if the other platforms you mentioned have (/ have had) the same problem.

gebart, 2018-05-15:
I would like to remove it, but I don't have time to work on it.

closed 2019-09-10 by stale[bot]