RIOT-OS/RIOT #4882: Test for random number generation distribution missing

gebart, 2016-02-23:
It would be interesting to see if [TestU01](http://simul.iro.umontreal.ca/testu01/tu01.html) would be possible to run on RIOT. BOARD=native would be a good starting point, but running the test on the hardware random number generators on the different boards would also be an interesting test.

kaspar030, 2016-02-23:
My entropy collector PR (#4294) includes a test harness that can run tests on numbers generated by our prng's in a mostly automated way, but with the code running on the host system, so it wouldn't work for the hwrngs.

kaspar030, 2016-02-23:
I'll look into adding TestU01.

basilfx, 2017-11-27:
I've spend an hour to see how TestU01 could be used, but it requires serious patching before it even compiles (missing externs, lots of errors). @kaspar030, my guess is that you didn't continue either?

Maybe, for on-target testing, we could use something simple (but commonly used) FIPS 140-2 testing (like used in [rngtest](https://linux.die.net/man/1/rngtest)). That would be easy enough to re-implement from scratch (there are existing sources, but GPL).

Shall I look into that?

PR #8182 (2017-12-18): tests: rng: add test tools for RNG sources
This PR adds {HW, P}RNG test tools. It was part of #5675. Current tests include:

* Distributions
* Speed
* Entropy
* FIPS 140-2

It should (partially) fix #4882 and check-off a box in #7886.

I've included pexpect test that uses the default PRNG and a constant source (for testing failures).

This test tools doesn't guarantee that a PRNG is safe/secure/good to use, but it should validate its response and provide you the tools to play with your PRNG. For improved statistical tests, you can use the `dump` command to dump numbers and use offline tests su [... cut]
closed 2017-12-18 by basilfx