RIOT-OS/RIOT #4893: Board z1: wake up from LPM

Joelfp, 2016-02-24:
I've been testing the LPM library (lpm.h) with the zolertia board. As far as I understand the lpm_set(lmp_mode_target) sets the board to one of the 5 LPM of the MSP430 processor. This works fine. But when using lmp_awake() inside an interrupt function I cannot get the board to wakeup again.
How am I suppose to use this library correctly?

malosek, 2016-02-25:
Hello,

not sure if I understand your use case, but have a look at gpio_init_int where you can register callback.
LPMs should be managed by the OS core not by the user and MCU should be in the sleep mode where there is no task to be executed..

wbr
malo

Joelfp, 2016-03-10:
As I understand, during the sleep mode the CPU stops executing instructions until an external interrupt is executed. For my project I requiere to put the cpu in sleep mode to save as much energy as posible.
In the code I use gpio_init_int to create a callback function for the interruption and then set the lpm mode with
`lpm_set(lmp_mode_target);`
In the callback function I use the function
`lpm_awake();`
to wake the cpu from the sleep mode. But when I execute and external interrupt from the user button I enter the callback function but the cpu does not wake up.
On the other hand when I implement the sleep mode with the defines provided for the MCU I can enter and exit the sleep mode correctly.
This is my question: If I am using incorrectly the lpm library to wake up the cpu from the sleep mode?

PeterKietzmann, 2017-11-27:
@Joelfp did you follow recent discussion on the mailing list and PRs related to power management in general? Couple of things have changed and this issue is actually not up to date I guess. Mind to close?

PeterKietzmann, 2018-01-31:
This topic is outdated and seems to lack attention. Closing this one

closed 2018-01-31 by PeterKietzmann