RIOT-OS/RIOT #4898: Warning example gnrc_networking + libcoap

bapclenet, 2016-02-25:
I tried to simply compile the sample gnrc_networking + the package libcoap (without editing something in main.c), I receive:

```
/home/xsoen/Documents/Developpement/workspace/RIOT-2016.02/sys/posix/sockets/posix_sockets.c:184:12: error: '_implicit_bind' defined but not used [-Werror=unused-function]
 static int _implicit_bind(socket_t *s, void *addr)
            ^
cc1: all warnings being treated as errors
```

bapclenet, 2016-02-25:
By the way, I tried to make the server libcoap example and I get errors, has it been already compiled by someone?

OlegHahm, 2016-02-25:
`tests/coap` which also uses libcoap builds without warnings or errors for native and samr21-xpro.

OlegHahm, 2016-02-25:
> By the way, I tried to make the server libcoap example and I get errors, has it been already compiled by someone?

Which example?

bapclenet, 2016-02-25:
Example which are provided by libcoap (`libcoap/examples/server.c`)
Ok `tests/coap` works but we don't use anything from libcoap inside

OlegHahm, 2016-02-25:
if `libcoap/examples/server.c` fails to build this should be reported in the libcoap issue tracker, I guess.

bapclenet, 2016-02-25:
libcoap issue tracker?
Ok I change the name of the issue because libcoap compile in the test

OlegHahm, 2016-02-25:
Well, I think we have two different issues here:
1.) adding `USEPKG += libcoap` result in the above mentioned warning -> error.
2.) `libcoap/examples/server.c` fails to build.

1.) is a problem with RIOT's dependency resolution in Make. I will create a PR to fix this.
2.) is a problem related to libcoap and should be reported to the libcoap maintainers.

OlegHahm, 2016-02-25:
Here you go: #4900.

bapclenet, 2016-02-25:
@authmillenon, you did the port of libcoap to RIOT, did you try to use it or you only compiled it?

OlegHahm, 2016-02-25:
I used it last November, but I couldn't find my example application.

bapclenet, 2016-02-25:
Let me know if you find it, I want to to see how much RAM and ROM it takes and try the OBSERVE capability.
I see three implementations of coap for RIOT:
- microcoap: server side without OBSERVE
- libcoap: client and server side with OBSERVE (use apparently more memory than microcoap)
- Soletta Project: client and server side with OBSERVE but totally dependant of Soletta middleware

The best would have been microcoap with OBSERVE capability :-)

OlegHahm, 2016-02-25:
I think @kb2ma is also workling on a CoAP implementation specifically for RIOT and GNRC, but I don't know the current state.

kb2ma, 2016-02-25:
Thanks for the mention, @OlegHahm! You can see the current state of the work in my [gcoap/client-2015.12](https://github.com/kb2ma/RIOT/commits/gcoap/client-2015.12) branch. It is focused now on client side communication rather than server side. It can send a simple non-confirmable message.

This work is an "in my free time" project. I plan to get back to it, but don't have a schedule.

bapclenet, 2016-02-25:
@kb2ma, thanks for you work, are you going to implement the OBSERVE capability for gnrc_coap?

kb2ma, 2016-02-25:
I think Observe is a sensible way to implement a push capability. It is on my wish list for the project, but I don't have a timetable for implementation. 

@bapclenet, if you'd like to take what I have done and develop it further, or just use it for inspiration, please feel free. If I get time to collaborate, I would be happy to. I think it's more important that the work gets done than anyone in particular does it. :-) Please send a link if you decide to work on it.

bapclenet, 2016-02-25:
Or we could adapt libcoap implementation for RIOT as they did for contiki with `#ifdef WITH_CONTIKI` then libcoap would be gnrc compliant and not POSIX
At the moment I can't make a libcoap example works. @OlegHahm, your example application would be welcome :-)

@kb2ma, I will think about it and see which is the best/quickest way to get OBSERVE for my app

authmillenon, 2016-02-25:
> @authmillenon, you did the port of libcoap to RIOT, did you try to use it or you only compiled it?

I ported it years ago, and it was used (and changed) multiple times since then. So I'm not sure if I'm still the most suitable for this task. I reassign @OlegHahm since he already opened a PR to fix it.

closed 2016-02-25 by bapclenet

bapclenet, 2016-02-25:
I reset the previous name as the PR fixed this problem and I created a new PR for the libcoap server example 
https://github.com/RIOT-OS/RIOT/issues/4901

PR #4900 (2016-02-25): libcoap: fix missing dependencies and includes
fixes #4898.