RIOT-OS/RIOT #4901: libcoap: server.c example does not compile

bapclenet, 2016-02-25:
Example server in pkg/libcoap/examples doesn't compile with RIOT

OlegHahm, 2016-02-25:
Can you tell how you tried to build this with RIOT?

Actually, I still think you should open this issue in the issue tracker of the libcoap project.

bapclenet, 2016-02-25:
I used a quick and brutal way:
Delete main.c in gnrc_networking, copy server.c and rename it to main.c, add libcoap as a pkg in the Makefile and compile
It doesn't compile (I didn't really expect it to compile this way). 
RIOT should provide an example of each pkg it works with.

OlegHahm, 2016-02-25:
> RIOT should provide an example of each pkg it works with.

I don't know if I agree with this.

bapclenet, 2016-02-25:
>I don't know if I agree with this.
IMO, if someone brings a package to RIOT, it should somewhere show the most simple things you can do with it but that's my point of view :-)

OlegHahm, 2016-02-25:
Ok, sorry, for my brief response. Yes, it should be demonstrated somewhere (tests, application repository). I would just keep the example list rather short. IMO the examples should demonstrate rather the main capabilities of RIOT than containing an application for every module and package. Since we already have one coap example (based on microcoap), I think that's enough.

bapclenet, 2016-02-25:
>Yes, it should be demonstrated somewhere (tests, application repository)

/tests would be a good one.

>IMO the examples should demonstrate rather the main capabilities of RIOT than containing an application for every module and package

You're definitely right for that!

>Since we already have one coap example (based on microcoap), I think that's enough

IMO, no because it doesn't implement all the coap capabilities or at least the OBSERVE capability which is IMO required for a coap server, it avoids to pull frequently a resource to see if it has change.

aabadie, 2017-07-07:
gcoap/nanocoap is besoming the reference coap implementation in RIOT: it now handles OBSERVE request.
Maybe it is worth trying this.

Closing this one but feel free to reopen if you disagree.

closed 2017-07-07 by aabadie