RIOT-OS/RIOT #4920: travis: using old configuration with current scripts

OlegHahm, 2016-02-28:
The discussion in https://github.com/RIOT-OS/RIOT/pull/4389#issuecomment-189715329 revealed that we have problem with older PRs if the Travis configuration and/or the test scripts change. One possible solution that I see, could be by parameterizing the `build_and_test` script with the branch name of the current build and let Travis run the version of the script that belongs to the currently checked PR.

authmillenon, 2016-02-28:
The problem why this arises is [this rebase](https://github.com/RIOT-OS/RIOT/blob/master/dist/tools/travis-scripts/build_and_test.sh#L31). Travis usually creates a merge commit with the current merge base *on build creation* (which means it doesn't change if the build is just restarted) so this rebase is most likely only there to catch errors that happen due to changes in master *after build creation*

authmillenon, 2016-02-28:
Removing this rebase would result in exactly the behavior you proposed on the expense of breaking master due to errors that the version of s PR just wouldn't pick up because it uses an old version of master.

authmillenon, 2016-02-28:
I think I have an idea how to fix that, but first I go watch Deadpool now ;)

authmillenon, 2016-02-29:
The fix is now available at #4921 btw ;-)

PR #4921 (2016-03-03): travis: warn about differing .travis.yml
Fixes #4920 in the most simplest way: just warn about the differing .travis.yml.
closed 2016-03-03 by OlegHahm