RIOT-OS/RIOT #4927: dist/tools/ethos: ignoring return value of write

gebart, 2016-02-29:
The following warnings show up when building ethos in master:
```
dist/tools/ethos (master|✔)(328/251)% make
cc -O3 -Wall ethos.c -o ethos
ethos.c: In function ‘main’:
ethos.c:320:33: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
                                 write(tap_fd, serial.frame, serial.framebytes);
                                 ^
ethos.c:323:33: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
                                 write(STDOUT_FILENO, serial.frame, serial.framebytes);
                                 ^
ethos.c:356:13: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
             write(serial_fd, &delim, 1);
             ^
ethos.c:358:13: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
             write(serial_fd, &delim, 1);
             ^
ethos.c:371:17: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
                 write(serial_fd, head, sizeof(head));
                 ^
ethos.c:373:17: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
                 write(serial_fd, &delim, 1);
                 ^
ethos.c: In function ‘_send_hello’:
ethos.c:249:5: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
     write(serial_fd, head, sizeof(head));
     ^
ethos.c:250:5: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
     write(serial_fd, serial->local_l2_addr, 6);
     ^
ethos.c:251:5: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
     write(serial_fd, &delim, 1);
     ^
ethos.c: In function ‘_clear_neighbor_cache’:
ethos.c:258:5: warning: ignoring return value of ‘system’, declared with attribute warn_unused_result [-Wunused-result]
     system(tmp);
     ^
ethos.c: In function ‘_write_escaped’:
ethos.c:230:17: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
                 write(fd, out, 2);
                 ^
ethos.c:236:17: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
                 write(fd, out, 2);
                 ^
ethos.c:240:17: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
                 write(fd, &c, 1);
                 ^
```
Version info:
```
% cc --version
gcc (Gentoo 5.3.0 p1.0, pie-0.6.5) 5.3.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

OlegHahm, 2016-02-29:
This looks like a wontfix to me. `-Wunused-result` would lead to a lot of `(void) do_foobar()` lines if we handle it consequently.

kaspar030, 2016-03-10:
@gebart What libc is gentoo using? My arch doesn't complain. Is that function really declared with that attribute?

haukepetersen, 2016-04-13:
I have the same with `gcc (Ubuntu 4.8.4-2ubuntu1~14.04.1) 4.8.4`... Though it's not critical, I think it adds to the user experience (expecially for newbies), if they can compile `ethos` without warnings... Don't care how we make that happen :-)

OlegHahm, 2016-04-13:
@gebart, @haukepetersen, can you check if #5305 eliminates the warnings?

haukepetersen, 2016-04-13:
did, no improvements for me...

haukepetersen, 2016-04-14:
what do you say to #5317?

haukepetersen, 2016-04-20:
fixed with #5317

closed 2016-04-20 by haukepetersen