RIOT-OS/RIOT #4954: chronos: compiling with -O0 breaks

cgundogan, 2016-03-03:
Compiling with `-O0` instead of `-Os` produces
```
RIOT/cpu/cc430/cc430-adc.c: In Funktion »adc_isr«:
RIOT/cpu/cc430/cc430-adc.c:113:48: Error: frame allocation destroys caller register due to »task« [-Werror=attributes]
```
Seems like `-Os` optimizes the frame away and therefore the error is not encountered.

see also #4413

Kijewski, 2016-03-03:
Please try
```diff
-__attribute__((naked))
+__attribute__((naked, optimize("omit-frame-pointer"), no_instrument_function))
```

cgundogan, 2016-03-03:
@Kijewski your patched seemed to work. I had to apply the same for `port1_isr()`, but now I am hitting
```
RIOT/cpu/cc430/cc430-gpioint.c:147:38: Error: function requires 8 bytes for stack storage but frame allocation inhibited by »naked« [-Werror=attributes]
```

Kijewski, 2016-03-03:
The problem is use of `xtimer_now()`, which calls [`timer_read()`](https://github.com/RIOT-OS/RIOT/blob/c8829a49b8263acd00852aaee716812d8bf709c0/cpu/cc430/periph/timer.c#L98). Being defined in another compiling unit, GCC cannot tell that the function is simply a lookup `(unsigned int)TIMER_BASE->R`, so it needs to assume that the call destroys its registers.

Try to manually inline the `xtimer_now()` or test  if the use of `((naked))` is needed at all (looks like a needless micro optimization).

DipSwitch, 2016-03-14:
Any luck?

cgundogan, 2016-03-23:
I currently have no access to any msp board. Does anyone volunteer to test?

OlegHahm, 2016-03-23:
If no-one volunteers, I can provide the hardware next Tuesday for the H&A session.

cgundogan, 2016-03-23:
>If no-one volunteers, I can provide the hardware next Tuesday for the H&A session.

+1

authmillenon, 2016-04-20:
Did that happen?

authmillenon, 2016-04-20:
Possibly fixed by #5188 (though the state of that PR is unclear)

kYc0o, 2016-07-11:
Can anyone check if #5188 solves this?

kYc0o, 2016-07-26:
#5188 was postponed so this one too.

PeterKietzmann, 2019-02-21:
This issue seems horribly outdated, and so does the chronos por :/. However, could anyone say (i) which application was compiled and (ii) how the `CFLAG` was set?

miri64, 2019-02-21:
I can still reproduce with

```sh
BOARD=chronos CFLAGS_OPT=-O0 make -C examples/default/
```

miri64, 2019-02-21:
(I amended OP to better document this)

PeterKietzmann, 2019-02-21:
merci

aabadie, 2020-10-29:
chronos support was removed in ed3887ac5c1e95308c2827bce3cdca8b0f146c22 (#12457) so this no longer applies I think.

bergzand, 2020-10-29:
Lets close this then. We can always reopen.

closed 2020-10-29 by bergzand