RIOT-OS/RIOT #4958: Does RIOT have an assertion macro similar to this one?

x3ro, 2016-03-04:
I did not find one, but think this could be super useful, possibly with an additional message? Right now assertions just result in kernel panics, it seems.

    #define assert(condition) do { \
            if(!(condition)) { \
                fprintf(stderr, "Assertion failed %s in file %s line %d\n", #condition, __FILE__, __LINE__); \
                abort(); \
            } \
        }

kaspar030, 2016-03-04:
We've replaced above assertion with a simple panic as every assertion would enlarge the binary by the length of the message in bytes + condition checking + function call, adding up to about ~120b per assertion.
Modules like the whole mutex code are maybe 500 bytes. Imagine the effect of a dozen assert()s on the code size...

x3ro, 2016-03-04:
Understandable, but I don't think there's a reason to not have this for debugging. If I'm not mistaken, basically the same thing regarding memory can be said for the `DEBUG` macro as well.

kaspar030, 2016-03-04:
Probably an #ifdef'ed including of the C library's <assert.h> would serve your needs.

DEBUG is a little different:
- it has to be enabled manually per-module
- there are a lot less DEBUG() statements than asserts
- DEBUG() statements tend to be shorter than asserts

x3ro, 2016-03-04:
You mean an ifdef'd include of the existing assert.h?

Kijewski, 2016-03-04:
I could think of using the same pattern as in `ENABLE_DEBUG`:

```c
#define VERBOSE_ASSERT (1)  // where VERBOSE_ASSERT==0 if definition absent
#include "assert.h"
```

x3ro, 2016-03-04:
Wouldn't it make sense to just let `ENABLE_DEBUG` enable verbose asserts as well?

OlegHahm, 2016-03-06:
Well, the current minized assert macro gives you the address of the failing assertion and just connecting a debugger shows you its pace very easily. If you want to use a verbose version of assert, the simplest way is to remove `core/include/assert.h` from your copy of the repo and fall back to Newlib's assert.h. I don't think that it is a good idea to add this huge amount of memory overhead for a rather small improvement to the convenience. Note that some boards are really limited in terms of ROM and by adding this big `assert()` again, you might not be able to fit the application at all.

miri64, 2016-10-18:
Fixed in #5881.

miri64, 2016-10-31:
#5881 got merged

closed 2016-10-31 by miri64