RIOT-OS/RIOT #4973: thread_t is already defined by OS X

Yonezawa-T2, 2016-03-07:
Compiling native platform on OS X (10.11.3) results in the following error:

```
~/src/RIOT/core/include/sched.h:98:24: error: typedef redefinition with different types
      ('struct _thread' vs 'mach_port_t' (aka 'unsigned int'))
typedef struct _thread thread_t;
                       ^
/usr/include/mach/mach_types.h:119:22: note: previous definition is here
typedef mach_port_t             thread_t;
```

#4578 renames `tcb_t` to `thread_t`.

Dependency graph:

`cpu/native/periph/timer.c → mach/mach_host.h → mach/mach_types.h`

Possible fixes:

- rename to `tcb_t`
- rename to `riot_thread_t`
- rename to `thrd_t`
- rename to `process_t`
- `#define thread_t riot_thread_t` in sched.h
- other?

LudwigKnuepfer, 2016-03-07:
Just a thought: isn't it possible to use OSX Jenkins installations as well? I guess this could have been caught... Other than that: I'd just kick support for OSX, let them use vagrant or something.

kaspar030, 2016-03-07:
@emmanuelsearch Could you check milhouse? I can't reach it... Is it still running?

kaspar030, 2016-03-07:
> other?

factor out mac os functions into a different compilation unit...
I'm reluctant to give up the name "thread_t" for mac os :)

Kijewski, 2016-03-07:
Wouldn't `#define thread_t riot_thread_t` in the offending .c file, just after the Mach includes, solve the problem? Other than that: what Ludwig said.

OlegHahm, 2016-03-07:
Which Jenkins?

LudwigKnuepfer, 2016-03-07:
Jenkins, Travis, what's the difference...

LudwigKnuepfer, 2016-03-07:
(Just to be clear: I meant Travis and typed Jenkins instead.)

Yonezawa-T2, 2016-03-09:
> Wouldn't #define thread_t riot_thread_t in the offending .c file, just at the end of the Mach includes, solve the problem?

It seems to be working, at least compiler is happy. `timer.c` does not use `thread_t`.

Yonezawa-T2, 2016-03-14:
Created PR #5056.

PR #5056 (2016-03-16): native: workaround for thread_t name collision
Fixes #4973.
Implements @Kijewski's idea.
closed 2016-03-16 by authmillenon