RIOT-OS/RIOT #4974: CI: TODO list

kaspar030, 2016-03-07:
- [ ] create nice webview for https://ci.riot-labs.de/api/pull_requests
- [ ] adapt pr_checks.sh labels check
- [ ] document the system

I won't have time, so please feel free and pick a task! :)
I'll help where I can.

I'll try to tackle these:
- [ ] distribute builds
- [ ] optimize build scripts

DipSwitch, 2016-03-07:
Don't know if I've time. But to help the process speed-up. What would be
the requirements for the web interface?

- Pending jobs
- Job results
- (Re)queue / run
- Mail developer on failed run (not really web)

Suggestions are more then welcome :)

kaspar030, 2016-03-07:
I'll copy your post into a new issue (CI UI wishlist).

authmillenon, 2016-03-07:
Not building commits marked with `[ci skip]` ;-)

authmillenon, 2016-03-07:
From #4645:

@authmillenon
>Another thing is that I marked the last commit to be skipped by CI... RIOT CI ignores it.

@kaspar030
> I thought you were joking. Uncheck "ready for ci".

@authmillenon
> But that skips the build for the PR altogether...

@kaspar030
> So you're waiting for non-head build results?

Yes. If I do a small doc change I don't want to restart the CI all over.

kaspar030, 2016-03-07:
Discussion from #4645.
RIOT CI cancels all but the most current builds for a PR.

kaspar030, 2016-03-07:
> So you're waiting for non-head build results?
> 
> Yes. If I do a small doc change I don't want to restart the CI all over.

Why not?

kaspar030, 2016-03-07:
Any change has to be re-checked, anyways.

authmillenon, 2016-03-07:
(In the case of #4645 it was no doc change but the `[ci skip]` label was added by my git hooks)

authmillenon, 2016-03-07:
> Why not?

> Any change has to be re-checked, anyways.

1. Does RIOT CI check spelling errors?
2. It does so to as soon as I squash...

Though RIOT CI is faster then Travis we shouldn't overload it either so way not add this feature for conscious contributors.

kaspar030, 2016-03-07:
> Though RIOT CI is faster then Travis we shouldn't overload it either so way not add this feature for conscious contributors.

It doesn't help if the (outdated) build continues, as it hogs the queue. So keeping it in, knowing that another commit has to be checked, soon, is something a conscious contributor shouldn't do.

authmillenon, 2016-03-07:
Then why not stop RIOT CI completely if HEAD contains `[ci skip]`?

authmillenon, 2016-03-07:
(mind you: not everyone can set label, but a good contributor knows which commits are important to test and which not)

kaspar030, 2016-03-07:
Thing is, if you're using RIOT CI to test your build, even if you expect failures, that is fine. But that should not cost too many resources.
If we continue building non-head builds, especially for those contributors who keep hammering out commits, every PR is suddenly multiple times in the build queue. That is suboptimal.
If you want a build to complete, don't commit. If you want a commit not to be built, uncheck "ready for ci".

Canceling a running build, but ignoring head, if a commit with "ci skip" comes in, would be an option.

authmillenon, 2016-03-07:
As I said: not every contributor can set/unset the labels. Just the maintainers. `ci skip` would give (knowledgeable) contributors the power to stop a running build.

kaspar030, 2016-03-07:
ok, and easy to do. I'll add it to the list.

OlegHahm, 2016-03-08:
It seems that there's another problem here: https://github.com/RIOT-OS/RIOT/pull/4983#issuecomment-193745612

Kijewski, 2016-03-08:
@kaspar030, could you add the header lines
```
Access-Control-Allow-Credentials: false
Access-Control-Allow-Origin: *
```
to https://ci.riot-labs.de/api/pull_requests so one can easier play with it?

kaspar030, 2016-03-08:
@Kijewski done. That basically allows cross-site-scripting, right? Remind me to turn it off at some point. :)

Kijewski, 2016-03-08:
@kaspar030, exactly. `Access-Control-Allow-Origin: *` allows to fetch the page via ajax from any origin (~domain). `Access-Control-Allow-Credentials: false` commands the browser not to transmit cookies (not sending cookies is the default, but explicitly denying it does not hurt).

OlegHahm, 2016-03-09:
The build in https://github.com/RIOT-OS/RIOT/pull/5008 was failing with
```
Assembler messages:
Fatal error: can't create ../../../../../../tests/coap/bin/stm32f0discovery/gnrc_ipv6_nc/gnrc_ipv6_nc.o: No such file or directory
/data/riotbuild/Makefile.base:60: recipe for target '/data/riotbuild/tests/coap/bin/stm32f0discovery/gnrc_ipv6_nc/gnrc_ipv6_nc.o' failed
make[4]: *** [/data/riotbuild/tests/coap/bin/stm32f0discovery/gnrc_ipv6_nc/gnrc_ipv6_nc.o] Error 1
/data/riotbuild/Makefile.base:20: recipe for target 'ALL--/data/riotbuild/sys/net/gnrc/network_layer/ipv6/nc' failed
make[3]: *** [ALL--/data/riotbuild/sys/net/gnrc/network_layer/ipv6/nc] Error 2
make[3]: *** Waiting for unfinished jobs....
Assembler messages:
Fatal error: can't create ../../../../../tests/coap/bin/stm32f0discovery/gnrc_ipv6/gnrc_ipv6.o: No such file or directory
/data/riotbuild/Makefile.base:60: recipe for target '/data/riotbuild/tests/coap/bin/stm32f0discovery/gnrc_ipv6/gnrc_ipv6.o' failed
make[4]: *** [/data/riotbuild/tests/coap/bin/stm32f0discovery/gnrc_ipv6/gnrc_ipv6.o] Error 1
/data/riotbuild/Makefile.base:20: recipe for target 'ALL--/data/riotbuild/sys/net/gnrc/network_layer/ipv6' failed
make[3]: *** [ALL--/data/riotbuild/sys/net/gnrc/network_layer/ipv6] Error 2
/data/riotbuild/sys/net/gnrc/network_layer/ipv6/netif/gnrc_ipv6_netif.c:891:1: fatal error: opening dependency file /data/riotbuild/tests/coap/bin/stm32f0discovery/gnrc_ipv6_netif/gnrc_ipv6_netif.d: No such file or directory
 }
 ^
compilation terminated.
/data/riotbuild/Makefile.base:60: recipe for target '/data/riotbuild/tests/coap/bin/stm32f0discovery/gnrc_ipv6_netif/gnrc_ipv6_netif.o' failed
make[4]: *** [/data/riotbuild/tests/coap/bin/stm32f0discovery/gnrc_ipv6_netif/gnrc_ipv6_netif.o] Error 1
/data/riotbuild/Makefile.base:20: recipe for target 'ALL--/data/riotbuild/sys/net/gnrc/network_layer/ipv6/netif' failed
make[3]: *** [ALL--/data/riotbuild/sys/net/gnrc/network_layer/ipv6/netif] Error 2
/data/riotbuild/Makefile.base:20: recipe for target 'ALL--/data/riotbuild/sys/net/gnrc' failed
make[2]: *** [ALL--/data/riotbuild/sys/net/gnrc] Error 2
/data/riotbuild/Makefile.base:20: recipe for target 'ALL--/data/riotbuild/sys' failed
make[1]: *** [ALL--/data/riotbuild/sys] Error 2
/data/riotbuild/Makefile.include:218: recipe for target 'all' failed
make: *** [all] Error 2
Resolve operation not in progress, we are not resuming.
```

gebart, 2016-03-09:
Added link to "publish buildsizes after build (issue #3383)"

OlegHahm, 2016-03-09:
Another failing build that shouldn't have failed:
https://ci.riot-labs.de/RIOT-OS/RIOT/5004/edee25ced57251c8623c21c9e7680de6b07bd7a3/output.html

kaspar030, 2016-03-09:
Hm, they seem to have a similar problem: the folder doesn't get created. I'm investigating in #5018.

OlegHahm, 2016-03-09:
kk

OlegHahm, 2016-03-10:
Just found another fundamental shortcoming of the new CI: it doesn't have a proper name! I propose to name it "Knievel".

cgundogan, 2016-03-10:
>Just found another fundamental shortcoming of the new CI: it doesn't have a proper name! I propose to name it "Knievel".

Evel Knievel?

OlegHahm, 2016-03-10:
Exactly. It's fast, it's dangerous, and it lives on the edge. ;)

OlegHahm, 2016-03-10:
(and it never dies)

kaspar030, 2016-03-10:
nice! :) I was thinking of Murdock. The crazy, fun buddy of mr.t.

authmillenon, 2016-03-10:
How about Daredevil? It goes namewise in the same direction of you @kaspar030 (because of Matt Murdock) and reflects already the history of the CI ;-).

Kijewski, 2016-03-10:
> The crazy, fun buddy of mr.t.

I like Murdock.  :)

gebart, 2016-03-11:
+1 for Murdock, also fits nicely with "Docker"

gebart, 2016-03-11:
Suggestion: Add some static HTML links at the top of the build output to jump ahead to each build group summary. (and create the according HTML named anchors in the build output)

like: 
```
<a href="#mcu_group_cortex_m4_2">cortex-m4-2</a>
...
...
...
<a blah blah "mcu_group_cortex_m4_2" />
```
(I forgot the HTML syntax for named anchors, but you understand the idea)

kaspar030, 2016-03-11:
@gebart There are already the anchors. One has to klick on the "RESULTS" link, to jump to the actual build group links. I could make that the default link?

kaspar030, 2016-03-11:
Ah, I'm just realizing that there are only links for build groups with errors.

cgundogan, 2016-03-14:
@kaspar030 I am wondering: Travis skips tests that use files with no modifications. Murdock however seems to build everything for all PRs with no regard to the file modifications. If you are just using the compile_and_test python script, shouldn't it also skip some tests?

kaspar030, 2016-03-14:
@cgundogan ```build_and_test.sh``` starts compile_and_test with "$TRAVIS_BRANCH", which is empty for Murdock. Let's see if skipping makes things faster, I'll make a test PR.

kaspar030, 2016-03-14:
For the record: 2m28s! #5071.

authmillenon, 2016-03-17:
I know that was mentioned before somewhere, but since it still does not do that: Murdock should actually announce its real name in the build details on GitHub ;-)

authmillenon, 2016-03-17:
Is it me or is Murdock currently not picking up any PRs?

authmillenon, 2016-03-17:
Gateway timeout at https://ci.riot-labs.de/api/pull_requests

gebart, 2016-03-17:
The Murdock machine must be down...

Btw, how is it hosted? At FUB?

authmillenon, 2016-03-17:
AFAIK on a private server of @kaspar030 (with plans to move it to a server at FU).

kaspar030, 2016-03-18:
It's back now, it somehow lost ipv6 connectivity.

cgundogan, 2016-03-22:
I added a word about backtraces / core dumps

authmillenon, 2016-03-24:
Added word about reacting to label changes.

kaspar030, 2016-03-24:
> Added word about reacting to label changes.

Can you elaborate what's needed? The labels get checked at build-time. Do you want them to be checked when set?

authmillenon, 2016-03-24:
I want the build to be restarted (or at least the static test). E.g. in https://github.com/RIOT-OS/RIOT/pull/5134 the NEEDS SQUASHING label was removed a few seconds to late so that the builds overall failed, because the label check in static tests has failed (which is a double bummer, because the static tests are the first to be ready, but nobody can see that they failed until the build has finished).

kaspar030, 2016-03-24:
> I want the build to be restarted (or at least the static test).

This would be hacky to implement in murdock. Label changes get pushed asynchronously by github, so on every label change or at PR creation, murdock would have to start something like a "stable timer" to make sure all set labels have arrived. -> wontfix

Better fix the label check within our static tests.

authmillenon, 2016-03-25:
> Better fix the label check within our static tests.

What is there to fix? They do exactly what they should. Just that the labels are not set any more at the end of the whole test run.

kaspar030, 2016-03-25:
> What is there to fix? They do exactly what they should.

If they would, manual labeling "needs squashing" wouldn't have to fail or restart a build.

authmillenon, 2016-03-25:
But failing on the label is what the test is supposed to do. A restart would be difficult to do, since this could also solicit an infinite loop of rebuilds (if the "NEEDS SQUASHING" label is set because it is supposed to be, but the buildtests are supposed to run through)

kaspar030, 2016-03-25:
> But failing on the label is what the test is supposed to do.

And the label is supposed to be set if there are commits that need squashing. Improving the squash commit detection would make manually setting the label (and murdock's need to react on it) superfluous.

kaspar030, 2016-03-25:
But, if only failing a job with a set label is requested, that can actually easily be checked at the end of a build, avoiding the concurrency issues. Will look into it.

authmillenon, 2016-03-25:
> Improving the squash commit detection would make manually setting the label (and murdock's need to react on it) superfluous.

The test is already doing that. I think the label is more of a fallback, in case a contributor doesn't prefix the commit correctly. If we want to improve that we either need to enforce a commit naming scheme, that all contributors must follow or write a minor AI that recognizes the intend of a commit... I'm not willing to implement any of those solutions ;)

OlegHahm, 2016-03-29:
@kaspar030, I'm observing empty outputs from Murdock since yesterday, e.g. here https://ci.riot-labs.de/RIOT-OS/RIOT/3495/d72c2bee68845d55be956a82d8eb7cbb58e9c71a/output.html

cgundogan, 2016-03-29:
>@kaspar030, I'm observing empty outputs from Murdock since yesterday, e.g. here https://ci.riot-labs.de/RIOT-OS/RIOT/3495/d72c2bee68845d55be956a82d8eb7cbb58e9c71a/output.html

As a nasty workaround until it's fixed you can use the raw data with the extension `.txt` iso `.html`
https://ci.riot-labs.de/RIOT-OS/RIOT/3495/d72c2bee68845d55be956a82d8eb7cbb58e9c71a/output.txt

cgundogan, 2016-03-29:
>As a nasty workaround until..

It doesn't look soo nasty if `curl https://ci.riot-labs.de/RIOT-OS/RIOT/3495/d72c2bee68845d55be956a82d8eb7cbb58e9c71a/output.txt | less` (:

kaspar030, 2016-08-03:
I need to change the network connection of the buildbox, CI will be down for some minutes.

kaspar030, 2016-08-03:
> I need to change the network connection of the buildbox

done.

[11 more replies]

closed 2018-04-09 by kaspar030