RIOT-OS/RIOT #4982: RIOT CI webui wishlist

kaspar030, 2016-03-07:
Whishlist for probably coming RIOT CI web interface.

kaspar030, 2016-03-07:
Dipswitch wrote:
> Don't know if I've time. But to help the process speed-up. What would be the requirements for the web
 > interface? 
> - Pending jobs
> - Job results
> - (Re)queue / run 
> - Mail developer on failed run (not really web)
> Suggestions are more then welcome :)

kaspar030, 2016-03-07:
> Pending jobs
> Job results

Yes, basically a nice view of https://ci.riot-labs.de/api/pull_requests.

> (Re)queue / run

Yeah, and also cancel, for multiple PR's. Nice would be if the web interface would do oauth against github, so we don't need to deal with a user database. The browser side could manage all that and then just send a json rpc post with a list of PR's to restart/cancel, and the github username + auth token. Then it's supersimple to do on the server side.

> Mail developer on failed run (not really web)
I'm not sure github exposes the user email addresses. If yes, piece of cake. If not, we'd have to manage the github user -> email mapping somehow.

authmillenon, 2016-03-31:
I was fed-up with looking at JSON output so I wrote a simple viewing HTML: http://authmill.spline.de/pull_requests.html. Maybe someone wants to put that on the server. It's basically the JSON stuff in fancy bootstrap view.

It's my first JS coding in years so be nice ;-) (hope I handled the queued entries right since at the time of coding no queuing occured).

DipSwitch, 2016-03-31:
Looks good!

kaspar030, 2016-03-31:
@authmillenon Nice!

I made it available at https://ci.riot-labs.de/. If you feel like, you could PR it at https://github.com/kaspar030/murdock.

authmillenon, 2016-03-31:
My HTML file isn't on this repository ;-)

miri64, 2016-06-01:
Weeks later I realize

> I made it available at https://ci.riot-labs.de/.

Shouldn't that rather be https://ci.riot-labs.de/RIOT-OS/RIOT/

miri64, 2016-10-18:
Ping?

OlegHahm, 2017-01-15:
Do we still need this issue?

aabadie, 2017-07-07:
I think we can close this one. @kaspar030 ?

kaspar030, 2017-09-19:
Nope, I'll close.

closed 2017-09-19 by kaspar030