RIOT-OS/RIOT #4986: at86rf2xx: Same short address for multiple devices

gebart, 2016-03-08:
Since the driver only takes the topmost 16 bits of the cpuid, all my devices get the same short address initially because only some bits of the cpuid vary between the individual units. This does not affect the functionality because long address mode is the default, but still it feels like it should be hashed somehow to let all bits of the cpuid affect the address.

https://github.com/RIOT-OS/RIOT/blob/master/drivers/at86rf2xx/at86rf2xx.c#L125

OlegHahm, 2016-06-01:
Hm, IIRC IEEE 802.15.4 proposes that initially all devices should have 0xFFFF as short address and the PAN coordinator takes care of assignment. However, I think assigning some kind of pseudo random short addresses at boot-up seems to be somewhat sensible to me (especially since we do not support PAN coordinators). But I don't know how we can make sure that short addresses are somewhat uniq. @PeterKietzmann, @miri64, what was the conclusion from the latest discussion on this topic?

miri64, 2016-06-01:
See https://github.com/RIOT-OS/RIOT/issues/5457. Basically the current proposal is to refine the calculation process. However, this issue seems to be a duplicate of #5457 so close?

PeterKietzmann, 2016-06-02:
Yes. Short address generation is part of our discussions in #5457. Even if #5457 is more a duplicate of this issue :-) I will close it because there's been more discussions in the other one

closed 2016-06-02 by PeterKietzmann