RIOT-OS/RIOT #4987: native: pgrep: option requires an argument -- 'u'

gebart, 2016-03-08:
`make buildtest` shows the following message when building for native:

```
Building for native .. success
pgrep: option requires an argument -- 'u'

Usage:
 pgrep [options] <pattern>

Options:
 -d, --delimiter <string>  specify output delimiter
 -l, --list-name           list PID and process name
 -a, --list-full           list PID and full command line
 -v, --inverse             negates the matching
 -w, --lightweight         list all TID
 -c, --count               count of matching processes
 -f, --full                use full process name to match
 -g, --pgroup <PGID,...>   match listed process group IDs
 -G, --group <GID,...>     match real group IDs
 -i, --ignore-case         match case insensitively
 -n, --newest              select most recently started
 -o, --oldest              select least recently started
 -P, --parent <PPID,...>   match only child processes of the given parent
 -s, --session <SID,...>   match session IDs
 -t, --terminal <tty,...>  match by controlling terminal
 -u, --euid <ID,...>       match by effective IDs
 -U, --uid <ID,...>        match by real IDs
 -x, --exact               match exactly with the command name
 -F, --pidfile <file>      read PIDs from file
 -L, --logpidfile          fail if PID file is not locked
 --ns <PID>                match the processes that belong to the same
                           namespace as <pid>
 --nslist <ns,...>         list which namespaces will be considered for
                           the --ns option.
                           Available namespaces: ipc, mnt, net, pid, user, uts

 -h, --help     display this help and exit
 -V, --version  output version information and exit

For more details see pgrep(1).
```

Kijewski, 2016-03-08:
Should ${USER} contain the euid? Then 
```diff
-${USER}
+$$(id -u)
```

gebart, 2016-03-09:
I don't think it's correct that it is even running that pgrep command at all since it is supposed to be part of the debug-valgrind target. The Makefile should be changed IMO.
There's plenty more of these unnecessary extra shell calls in the makefiles which gets run even when not the intended make goal is selected

LudwigKnuepfer, 2016-03-09:
Since when?

cgundogan, 2016-03-09:
I can confirm @gebart's error message.

LudwigKnuepfer, 2016-03-09:
> Since when?

`2b949fcc boards/native/Makefile.include (Oleg Hahm          2016-01-26 17:38:15 +0100  92) debug-valgrind: export VALGRIND_PID ?= $(shell pgrep -n memcheck-x86-li -u ${USER} | cut -d" " -f1)`

OlegHahm, 2016-03-09:
@cgundogan, you're the expert here: how can we avoid that this line gets evaluated when it's not needed?

PR #5080 (2016-03-15): make: export VALGRIND_PID only for debug-valgrind
Not very pretty, but should fix #4987.
closed 2016-03-15 by cgundogan