RIOT-OS/RIOT #4993: doc: conn_udp: document that send and receives needs to be called from the same thread

OlegHahm, 2016-03-08:


authmillenon, 2016-03-08:
That's rather GNRC specific... so I don't think `conn_udp` is the right place for that.

authmillenon, 2016-03-08:
(also it's more a bug than a feature, so I would rather fix that instead of documenting faulty behavior ;-))

OlegHahm, 2016-03-08:
Well, it is currently the case and confuses users...

authmillenon, 2016-03-08:
I have to solutions in my sleeve:
* the fast and easy fix is to have thread running in `gnrc_conn.c` that acts as a supervisor for all connections and hands all received packets to the various registered connections (that's basically how I did it for emb6 and lwip, but I did not an extra thread since they are already single threaded and pretty much do all that for you)
* the more sophisticated and memory saving method would be to use #4919 for that and give every connection it's own mbox. This would require an adaption to `gnrc_netreg` to also take `mbox`s into account.

OlegHahm, 2016-03-08:
Since neither the one nor the other solution exists, I would rather have this documented.

authmillenon, 2016-03-08:
Will provide one as soon as I find the time.

PR #4995 (2016-03-08): doc: add notes that UDP receive functions must be called from same thread
fixes #4993.
closed 2016-03-08 by OlegHahm