RIOT-OS/RIOT #5005: ndp: router advertisement sent with global address

immesys, 2016-03-09:
I have spent a while trying to track down some weird behavior in my mesh. A few minutes after booting everything up, nodes start getting islanded off the mesh and their neighbor tables start emptying.

After some debugging and sniffing, I discovered that router advertisements were being received with global source addresses, which causes them to be dropped [here](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ndp/gnrc_ndp.c#L496) because the RFC says they must come from link-local addresses.

Today I got some time to dig deeper and modified the nodes to check that they are choosing the correct addresses. Basically I added this check [after line 579 of grnc_ndp_internal](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ndp/internal/gnrc_ndp_internal.c#L579)
```
    if (!ipv6_addr_is_link_local(src)) {
      disableIRQ();
      DEBUG("ndp internal: SOURCE NOT LOCAL\n");
      while(1);
    }
```

Sure enough, it triggered after a while. I probably won't get a chance to fix this soon, and I think this is what has been bugging other people like @kYc0o , so I thought I would report it. 

Some additional information, in case it is relevant. I am using RPL, and I have about 5 nodes with a border router. All of the nodes are sending and receiving traffic from the internet. They are positioned such that some of them need at least two hops to get in/out.

immesys, 2016-03-09:
Some more information, the src is coming as a parameter to the function. I'll try track down where it is coming from.

gebart, 2016-03-09:
Could you post a cloudshark dump which shows what it looks like when this happens? https://enterprise.cloudshark.org/upload/

immesys, 2016-03-09:
I'll get the cloudshark upload when I get into work tomorrow. I think I might need a little help on this one. Can someone find the error in my logic here?
- router advertisements come from [`gnrc_ndp_internal_send_rtr_adv`](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ndp/internal/gnrc_ndp_internal.c#L422)
- This is sometimes invoked with the `src` as NULL, in which case #5006 should fix that
- It is only invoked with a specific `src` from one place, in [gnrc_ndp_router](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ndp/router/gnrc_ndp_router.c#L120), but here it is in a block that checks that src is link local

Yet my preliminary testing with #5006 shows that there are still packets in the air with full global source addresses, that are being dropped in the same place I mention above.

Are there any other sources of router advertisement packets that don't go through `gnrc_ndp_internal_send_rtr_adv` and/or is #5006 implemented incorrectly? I will continue on this tomorrow, but as Germany is in a different timezone I figured I might ask and wake up to an answer :-)

authmillenon, 2016-03-09:
Hi, as stated in #5006: the source is global because the destination is global. Is that correct behavior? Since the hop limit of a RS must be 255 and is dropped if not it can be assumed that RSs always come from a neighbor so it might be correct behavior.

immesys, 2016-03-09:
The router advertisements are sent in response to router solicitations which (in my observations) have global sources, so my understanding is that we send the router advertisements to global addresses from link local addresses. Perhaps someone else should check the RFC?

immesys, 2016-03-09:
To clarify, the RFC is unambiguous about the source needing to be link local, but I guess if we knew the link local address of the source of the RS then we could also send the RA to a link local address. Do we necessarily know the LL addr for the RS source? I was thinking we didn't...

immesys, 2016-03-09:
Yeah, rereading RFC 4861 section 4.1, I think the correct behavior is to send to the source of the RS which may be global, from the link local address.

kYc0o, 2016-03-09:
Thanks @immesys for your debugging. Yes, this is exactly what I'm experiencing. My main question is: Why this happen "sometimes" or "after a while" and not always from the beginning?
My observations lead me to a some kind of race condition about the timeout to delete an address from the NC [here](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L261) which triggers [this](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/gnrc/sixlowpan/nd/router.h#L85). But this is only the reason to be deleted.

Then I can imagine the "misbehaviour" about the RA avoids to the global address to get in again on the NC. In my experience, link-local addresses get never erased, and I see on wireshark that neighbour sollicitations and advertisements using link-local addresses are sent just after global addresses are deleted.

kYc0o, 2016-03-09:
@gebart I'm trying to upload my dump to cloudshark but I have the error "Server responded with 0 code". I'm able to see the dump in wireshark. I stored the dump either as a pcapng and pcap but no one works. Any hints?

gebart, 2016-03-09:
@kYc0o I got the same error this morning when I tried to upload the dump for #5007. Possibly a server problem right now

gebart, 2016-03-09:
btw, did anyone investigate how other implementations do this? Linux, Windows, OSX etc. Do they all respond to RSs with RAs when configured as routers?

authmillenon, 2016-03-09:
> btw, did anyone investigate how other implementations do this? Linux, Windows, OSX etc. Do they all respond to RSs with RAs when configured as routers?

I did not, but why would'n they? That's what RSs are for: to solicit a RA. To my knowledge we are currently the only ones who are implementing [RFC 6775](https://tools.ietf.org/html/rfc6775) to the word. And that RFC is pretty clear IMHO that a router is supposed to send [RAs on received RSs](https://tools.ietf.org/html/rfc6775#section-6.3) (given that the rtr_adv flag is set of course).

immesys, 2016-03-09:
Here is an example bad router advertisement. In this case it's from my border router, but I have in the past seen them from all nodes in the mesh, given enough time. (Packet 593 above shows a different node doing it). As @kYc0o says it is a little random as to when it happens, but when it starts it normally keeps going, destroying the mesh.

[image]

immesys, 2016-03-10:
I did not manage to find this bug, after spending quite a bit of time on it. @authmillenon I would be grateful for some assistance in tracking this down. I can't keep a mesh up for more than fifteen minutes before this kills it. It should be simple to use #5006 to fix the problem, once we actually find where the troublesome address is coming from.

authmillenon, 2016-03-10:
Sorry, currently I'm really trying to focus on getting my thesis done. If there is no progress until April I will certainly throw myself on the problem, but for now I'm simply not able to :(.

immesys, 2016-03-10:
I understand, I have a paper deadline tomorrow too

authmillenon, 2016-03-19:
Not directly related to this issue, but I was able to set-up some tests with this this weekend and what I discovered was that the address registration for the GUA isn't repeated (leading to the address getting removed from the routers neighbor cache). I don't know why though so I have to investigate some more.

OlegHahm, 2016-03-19:
> I discovered was that the address registration for the GUA isn't repeated 

To clarify: on which layer should it be repeated? And how often? Is any RFC saying anything about this? And what do you mean by "address registration" here? The NS?

authmillenon, 2016-03-20:
> To clarify: on which layer should it be repeated? 

The Address Registration option is part of NDP so I guess you mean that?

> And how often?

Indefinitely, every few minutes "well before" the previous registration runs out.

> Is any RFC saying anything about this? 

https://tools.ietf.org/html/rfc6775#section-5.5.1

> And what do you mean by "address registration" here? The NS?

A NS/NA exchange with an ARO in it.

miri64, 2017-11-06:
Fixed by #7925

miri64, 2017-11-27:
Did not see this, when testing #7925 (which is now merged)

closed 2017-11-27 by miri64