RIOT-OS/RIOT #5007: gnrc icmpv6: Ping reply goes out the wrong interface

gebart, 2016-03-09:
When trying out the new border router setup with ethos in #4725 I found that if I run a ping on my linux machine to the address of the radio interface on the border router the reply is sent out the radio interface instead of sent back through the ethos pipe.

Cloudshark doesn't seem to want to cooperate with me right now so I can't upload the dump, but here's a screenshot of two Wiresharks side by side
[image]

The command on the Linux side was simply:
ping6 2001:16d8:ff00:8645:1016:4e54:8bab:4012

Linux net setup:
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 fd00:dead:beef::1/128 scope global 
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
...
8: sixxs: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1280 qdisc noqueue state UNKNOWN group default qlen 500
    link/none 
    inet6 2001:16d8:ff00:645::2/64 scope global 
       valid_lft forever preferred_lft forever
    inet6 fe80::14d8:ff00:645:2/64 scope link 
       valid_lft forever preferred_lft forever
139: riot0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 500
    link/ether 96:cf:69:cf:73:6e brd ff:ff:ff:ff:ff:ff
    inet6 fe80::94cf:69ff:fecf:736e/64 scope link 
       valid_lft forever preferred_lft forever
    inet6 fe80::1/64 scope link 
       valid_lft forever preferred_lft forever
```
`ip -6 r`
```
2001:16d8:ff00:645::/64 dev sixxs  proto kernel  metric 256  pref medium
2001:16d8:ff00:8645::/64 via fe80::2 dev riot0  metric 1024  pref medium
unreachable fd00:dead:beef::1 dev lo  proto kernel  metric 256  error -101 pref medium
fe80::/64 dev docker0  proto kernel  metric 256 linkdown  pref medium
fe80::/64 dev wlan0  proto kernel  metric 256  pref medium
fe80::/64 dev sixxs  proto kernel  metric 256  pref medium
fe80::/64 dev riot0  proto kernel  metric 256  pref medium
default via 2001:16d8:ff00:645::1 dev sixxs  metric 1024  pref medium
```

RIOT border router:
```
ifconfig
Iface  6   HWaddr: 40:12  Channel: 0  Page: 2  NID: 0x777
           Long HWaddr: 12:16:4e:54:8b:ab:40:12 
           TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
           AUTOACK  CSMA  MTU:1280  HL:64  6LO  RTR  RTR_ADV  IPHC  
           Source address length: 8
           Link type: wireless
           inet6 addr: ff02::1/128  scope: local [multicast]
           inet6 addr: fe80::1016:4e54:8bab:4012/64  scope: local
           inet6 addr: ff02::1:ffab:4012/128  scope: local [multicast]
           inet6 addr: 2001:16d8:ff00:8645:1016:4e54:8bab:4012/64  scope: global
           inet6 addr: ff02::2/128  scope: local [multicast]
           
Iface  7   HWaddr: 00:21:d3:60:5a:8d 
           
           MTU:1500  HL:64  RTR  RTR_ADV  
           Source address length: 6
           Link type: wired
           inet6 addr: ff02::1/128  scope: local [multicast]
           inet6 addr: fe80::221:d3ff:fe60:5a8d/64  scope: local
           inet6 addr: ff02::1:ff60:5a8d/128  scope: local [multicast]
           inet6 addr: ff02::2/128  scope: local [multicast]
           inet6 addr: fe80::2/64  scope: local
           inet6 addr: ff02::1:ff00:2/128  scope: local [multicast]
           inet6 addr: 2001:16d8:ff00:8645:221:d3ff:fe60:5a8d/128  scope: global
```

OlegHahm, 2016-03-09:
I don't have much time right now. Maybe @cgundogan or @kYc0o can take a look.

kYc0o, 2016-08-04:
I'll try to look for the next release.

miri64, 2016-11-08:
Is this still an issue? I did not see that in a while.

miri64, 2016-11-08:
Regardless, I just tested the setup (pinging the border router's WPAN interface address from the Linux side) and it did not seem to be an issue anymore. That is why I would deem this issue as resolved.

closed 2016-11-08 by miri64