RIOT-OS/RIOT #5024: Zolteria z1 terminal output

pdNor, 2016-03-09:
I can't get the z1 to print to the terminal. I tried to upload the default and hello-world example, but the z1 does not print anything to pyterm. If i switch to a different terminal program i can see that the z1 prints garbage to the terminal, I've tied with different baudrates but to no help. If i compile and upload the same examples  the arduino due everything works like a charm. Is anyone else experience the same problems with the z1?

OlegHahm, 2016-03-09:
Have you tried with latest master? I mean after https://github.com/RIOT-OS/RIOT/pull/4892 was merged? @Joelfp, @xav13er, #2109 was fixed for you, right?

OlegHahm, 2016-03-09:
@pdNor, I checked last week with my Z1 and pyterm and everything worked like charm.

@alignan,  any ideas?

pdNor, 2016-03-09:
Okay, my bad I pulled latest master and it has been fixed. Thanks for the quick reply.

closed 2016-03-09 by authmillenon