RIOT-OS/RIOT #5025: gnrc_rpl should populate the neighbor cache

gebart, 2016-03-10:
Upon receiving a RPL DIO containing a PIO, a new FIB entry is added for the GUA of the sender via the LL addr of the sender. However, the LL addr is not added to the neighbor cache, even though it obviously is reachable since we just received routing information from it.

This causes reachability issues if not using ND protocol on the remote node since the RPL router will send out NS requests but receive no replies and therefore be unable to forward packets destined for the remote GUA.

gebart, 2016-03-10:
@cgundogan Does the description make sense or should I elaborate?

cgundogan, 2016-03-10:
makes sense, thanks for the pointer. I had this thought once but looks like I've forgotten about that. I will look into it later in the afternoon.

gebart, 2016-03-10:
Would it make sense to populate the ncache from other sources as well, such as ICMPv6 echo replies from link-local addresses?

cgundogan, 2016-03-10:
I am not sure whether it is a good idea to populate the ncache from the reception of a DAO message. I remember again why I dropped this thought..
RPL states in the RFC that it requires bidirectional links to function properly. (6Lo-)ND is **one** way to get those bidirectional links, so that the ncache itself **only** contains entries that are bidirectional. Receiving a DAO is not a strong indication whether the link is bidirectional or not. DIO traffic is (mostly) multicast (and sparse) in the downwards direction, while DAO traffic is (mostly) unicast in the upwards direction. Due to the known problems of wireless technology (interference, etc.) and "enhancements" of link layers (e.g. to apply QoS for certain traffic patterns, multicast, unicast, etc.) the link can be **highly** asymmetric. Receiving a DAO does not mean that my DIO, which I send out e.g. 1 minute ago, has reached that same child.

gebart, 2016-03-10:
@cgundogan Okay, what are the options for populating the ncache if not using ND?

cgundogan, 2016-03-10:
I don't know (: It's an interesting use case that you have here. In the "best" case, RPL should only accept a node as a parent, if it's also in the neighbor cache. Currently we are not checking for that and this won't work for scenarios where you have no neighbor cache at all.
 
To summarize again so that I understand you correctly:
You have 2 nodes (A and B)
A is the RPL root and has a neighbor cache (+ (6Lo-)ND is running)
B is a normal rpl router who joins the DODAG of A and has no neighbor cache and no ND running.

A will always expect an entry in its neighbor cache before it can send out traffic to B,
but since B is not participating in the RS/RA-NS/NA exchange with A, A will never get an entry for B in its ncache. Is this the situation?

gebart, 2016-03-10:
@cgundogan I have the exact scenario that you describe

cgundogan, 2016-03-10:
Okay, I am currently working on an additional feature for my thesis that pulls the bidirectionality checking into RPL (and probably populates the ncache correctly then). This would address your use case nicely, but it's not "the standard way". Maybe @emmanuelsearch @OlegHahm @authmillenon @BytesGalore know of a solution for that?

cgundogan, 2016-03-10:
Regarding using ICMPv6 Ping requests/responses: AFAIK, 6Lo-ND needs the `SLLAO` option to know the link-layer address of the neighbor, in case the link-layer address is shortened by 6lowpan (@authmillenon ?)

authmillenon, 2016-03-10:
6LoWPAN does not touch the link-layer addresses. It uses them to compress IPv6 addresses. ;) The SLLAO is primarily there for ND proxying and in case the IP layer does not have access to the lower layer headers.

emmanuelsearch, 2016-03-10:
Maybe we should consider creating a STALE entry in the neighbor cache for neighbors heard of, but without bidirectional check at ND level yet?

OlegHahm, 2016-03-10:
Maybe it makes sense to discuss this on the mailing list to allow a broader audience to participate?

emmanuelsearch, 2016-03-10:
+1 @gebart do you launch a thread on devel?

gebart, 2016-03-10:
@emmanuelsearch done.

gebart, 2016-03-10:
https://lists.riot-os.org/pipermail/devel/2016-March/003687.html

miri64, 2016-10-18:
Is there any result to that discussion?

OlegHahm, 2017-01-15:
ping

DipSwitch, 2017-06-21:
This is now the case if I stand correct? I saw the neighbors when enabling RPL? @miri64 any thoughts?

miri64, 2017-06-21:
As soon as RPL is ported to the NIB (see #6325 at al.) this should happen automagically ;-). I wouldn't call it "filling the neighbor cache" though, because this also implies reachability confirmation and all this stuff, but when next hop information is in the NIB, it will be used for NUD (thus also making it part of the neighbor cache).

bergzand, 2018-04-16:
RPL is ported to NIB, can we close this issue?

miri64, 2018-04-17:
Yes, filling up the forwarding table now puts the next hop in the same structure as the neighbor cache. As already stated: it isn't technically *in* the neighbor cache, but at the memory it will be when added to the neighbor cache. I think however, that the NIB callbacks for state changes are not yet ported to RPL yet.

miri64, 2019-08-10:
> RPL is ported to NIB, can we close this issue?

Yes, see my follow-up comment.

closed 2019-08-10 by miri64