RIOT-OS/RIOT #5027: tests: (re)move tests/led

OlegHahm, 2016-03-10:
Looking at #4849: this test seems a bit pointless to me - at least as a test. I can understand how this application can be helpful, while porting to a particular board, but for anyone who wants to use it to test the functionality of the LED macros on a certain board, this is rather useless, because there's no way to tell, what exactly would be correct or faulty behavior. IMO we should move this application either to the application repository or just keep it in some private repository.

authmillenon, 2016-03-10:
On a similar note it might be good to finally unify the `LED_.*_(ON|OFF|TOGGLE)` macros. Basically every board defines the colors as they like to, making assurance of the portability of a app really hard (e.g. my energy consumption tests where I needed to deactivate the LEDs).

OlegHahm, 2016-03-10:
Different issue.

haukepetersen, 2016-03-11:
I have not hard feelings about moving it somewhere else. It is very useful when touching the LED macros, but you are right, that the results depend very much on the platform you test on - not ideal for a test...

haukepetersen, 2016-03-11:
or how about https://github.com/haukepetersen/RIOT/commit/9a9aae852dea72f1037db00f8637411185db1287

OlegHahm, 2016-03-12:
Looks much better (from description and result, not from the code ;)). @A-Paul, do you also like the proposal?

haukepetersen, 2016-03-14:
I know, the code is ugly, but does the job...

A-Paul, 2016-03-16:
@OlegHahm and @haukepetersen  I have been *off* for some days and didn't know that you already addressed this issue.

> do you also like the proposal?

I ran the new (now merged) test. In my opinion, the console text and the behaviour of the LEDs is comprehensible now. :+1: 

> I know, the code is ugly, but does the job...

If in doubt, "form follows function". :-)

I also added a late checkmark for *cc2538dk* in #5045.

haukepetersen, 2016-03-17:
so can we close this issue?

OlegHahm, 2016-03-17:
I think so.

closed 2016-03-17 by OlegHahm