RIOT-OS/RIOT #5031: gnrc pktbuf: replace parameter checks with assert

OlegHahm, 2016-03-10:
Almost every function in `gnrc_pktbuf_static.c` seems to check its parameters for valid values. I guess we can replace most of these checks with asserts and save some ROM.

authmillenon, 2016-03-10:
This needs to be done very carefully. Sometimes a parameter might be "legally" in an invalid change, e.g. size == 0 if there is no payload for a packet.

OlegHahm, 2016-03-10:
> Sometimes a parameter might be "legally" in an invalid change, 

Can you rephrase, please?

authmillenon, 2016-03-10:
An assert lets the device effectively crash if there is an invalid value, while checking with an if and return an error code does not. There might be reasons why the parameters were wrong that might be totally feasible in the current program logic that use the `gnrc_pktbuf` API.

authmillenon, 2016-03-10:
(you don't save ROM if you have to check your all your parameter values before each call to `gnrc_pktbuf_add()` e.g., just do it in the function and return an error state)

OlegHahm, 2016-03-10:
May a null pointer also origin from a valid situation?

authmillenon, 2016-03-10:
For `gnrc_pktbuf_release()` e.g. I would say, yes. But for anything else I rephrase

> This needs to be done very carefully.

with "this" I mean which parameter states could be a legal state though invalid.

authmillenon, 2016-03-10:
And by "done" I mean "analyzed" :P

Yonezawa-T2, 2016-03-11:
Another concern is security. We should review the code to ensure no invalid parameters are passed to pktbuf even if malicious packets are injected.

closed 2019-09-10 by stale[bot]