RIOT-OS/RIOT #5034: gnrc_networking: offers txtsnd, does not print received packets

authmillenon, 2016-03-11:
`gnrc_networking` is offering the `txtsnd` command but does not print it results when receiving a packet. Overall this is a good situation (the application would otherwise get spammed with link layer packet output), but it might be confusing for new users. I'm voting to remove the `txtsnd` command from `shell_commands` an put it only in the applications were it is needed, as partly proposes in #2755.

OlegHahm, 2016-03-11:
Against removing txtsnd from shell commands. It's a very helpful debug tool.

authmillenon, 2016-03-11:
Even if there is no information printed at the receiver?

An alternative would be to introduce a dump command that would allow to activate the "dumpage" of a certain nettype.

OlegHahm, 2016-03-11:
Yes, even without - I can still see it in Wireshark - or in the debug output of my networking modules. ;)

OlegHahm, 2016-03-11:
To be clear: I see your general point and I second it.

haukepetersen, 2016-03-11:
I would not remove it, how about a `txtrcv` command, that waits exactly for 1 network packet to arrive and the prints this?

authmillenon, 2016-03-11:
You mean link layer packet. But this still might be confusing, since any network layer packet (Neighbor solicitations, RPL DIOs etc.) would also be printed and preempt the packet the user was originally expecting... I would opt more for a activatable/deactivatable solution.

OlegHahm, 2016-03-11:
Hm, difficult, particular, if you have additional traffic on the wire. On native, for instance, even the default example sees a lot of traffic on the link layer, stemming from the host system.

haukepetersen, 2016-03-11:
Isn't the `txtsnd` only compiled into the default example (without further network stack)?

authmillenon, 2016-03-11:
Nope, it's compiled into any application, that uses the `gnrc_netif` module.

haukepetersen, 2016-03-11:
ok, then forgot what I proposed :-)

OlegHahm, 2016-03-11:
But maybe you have a good point: how about making the txtsnd command optional and disable it per default? This would go into the direction of @authmillenon's initial proposal, but still make it available for debugging purposes if needed.

haukepetersen, 2016-03-11:
another dumb idea: how about to put a proprietary header (like `RIOT` in ASCII) in front of the data send via txtsnd. This could be easily parsed by the receiver...

haukepetersen, 2016-04-15:
hm, now that I though about his again: why don't we just drop `txtsnd` from the `gnrc_networking` example? If you want to focus on link layer only communication, use the default example instead?

OlegHahm, 2016-04-15:
Hm, doesn't this go back to the initial discussion?

haukepetersen, 2016-04-20:
won't be fixed for this release -> set Milestone to 2016.07

OlegHahm, 2016-06-01:
Hm, actually, I would be okay with dropping `txtsnd` from `gnrc_networking` example, but I'm still against removing the shell command. Anybody objects putting it into a separate module and include it only for `default` example?

miri64, 2016-06-01:
No objections.

kYc0o, 2016-07-26:
No objections, but please address it for the next release :)

LudwigKnuepfer, 2017-01-10:
Just to add some beef to the discussion which I did not read:
being able to txtsnd and with happening on the receiving side is very confusing.

OlegHahm, 2017-01-17:
\<theo-de-raadt-mode>Slackers!\</theo-de-raadt-mode>

OlegHahm, 2017-01-17:
see #6397

PR #6397 (2017-01-18): shell: introduce txtsnd pseudomodule
This commit allows to enable/disable the txtsnd shell command. The
command is used to send strings over L2 in GNRC. Until now the command
was automatically enabled if GNRC and shell_commands were present, which
may lead to confusion if no L2 packet handler is registered.

Fixes #5034.
closed 2017-01-18 by OlegHahm