RIOT-OS/RIOT #5037: new feature: netdev2_pcap

authmillenon, 2016-03-11:
Inspired by a device [existing in the lwIP contrib repository](http://git.savannah.gnu.org/cgit/lwip/lwip-contrib.git/tree/ports/win32/pcapif.h) I'm really intrigued by the idea of implementing a `netdev2` that just dumps and parses PCAP files to and/or from a stream (e.g. a UART or a file on native). These could be read by wireshark and would be link-layer independent while not needing adaption layers like ZEP or the 6LoENC ethertype.

It also would be great for testing, as we just could provide a well known PCAP file to such a device and see if the node reacts the expected way or vice-versa let it dump a PCAP file and check if it is as expected.

authmillenon, 2016-03-11:
Marked @OlegHahm as assignee to draw him to the discussion :P other opinions would be welcome, too. @BytesGalore @cgundogan @emmanuelsearch @gebart @haukepetersen @jfischer-phytec-iot @jremmert-phytec-iot @Lotterleben @kaspar030 @kYc0o @PeterKietzmann @zhuoshuguo ...

kaspar030, 2016-03-11:
Nice way for testing stuff!
Could use "xxd" or something to embed a .pcap into a testcase.

OlegHahm, 2016-03-11:
If it works in both direction (i.e. also outputting pcap) we could also get rid of the sniffer script, although I'm skeptical this can be done on-the-fly.

In general :+1: for the idea, but I think the next release is unrealistic.

haukepetersen, 2016-03-11:
I also like the idea, and when looking at the [PCAP format](https://wiki.wireshark.org/Development/LibpcapFileFormat) this does not seem too difficult.

authmillenon, 2016-03-11:
Yeah, that's why I opened an issue first (the milestone I set more or less in auto-pilot ^^) and not provided a solution. I think this will need some longer work.

authmillenon, 2016-03-11:
The PCAP stuff isn't to complicated, but the way how to handle IO needs to be thought over.

haukepetersen, 2016-03-11:
true

authmillenon, 2016-03-30:
Provided proof of concept implementation in #5206

miri64, 2016-10-18:
#5206 was disconitnued, but maybe someone else likes to pick up this task?

miri64, 2020-06-30:
This can basically be done with [`scapy`](https://scapy.readthedocs.io/en/latest/usage.html#reading-pcap-files). I don't think this needs to be integrated into RIOT anymore.

closed 2020-06-30 by miri64