RIOT-OS/RIOT #5046: DELAY_CHANGE_TXRX_US time for nrf24l01 double to make it work

srisler, 2016-03-11:
Hi guys,

Just to let you know, I had a code running on a STM32F3Discovery hooked with a NRF24L01P that used to work.

I updated RIOT (rebase on master) and it appears to fail to send messages (about 1 message over 20 went through and most often corrupted). I doubled the time of DELAY_CHANGE_TXRX_US and it works like a charm.

I still not have explanation as the value of DELAY_CHANGE_TXRX_US is correct according to the datasheet. The receiver is based on a STM32F0 (custom hardware, running on internal quartz for main clock), and has been updated the same way and managed to work without this change.

Does someone has an idea and if there could be a bug somewhere?

gebart, 2016-03-11:
Did you verify that the clocks on both devices run at the correct speed? You can use the `tests/xtimer_drift` test application to test this.

PeterKietzmann, 2016-03-15:
Hi @srisler looking to the latest changes to the driver I don't think your problems are related to those. Did you try @gebarts proposal? However, I need to test it but probably won't find time until the next Hack'n'ACK on 29.03

srisler, 2016-03-15:
Hi,

I am sorry, i have no time to test it this week. I'll come back to you with
update hopefully this week end.

Have a good day

srisler, 2016-03-20:
Hi,

Actually it does not work neither. I have some trouble make the NRF works with xtimer (apparently the only diff with previous code). 

I close this issue and open another one if needed.

FYI I ran the xtimer_drift test here is the result. It looks ok isn't it?:

```
2016-03-20 13:37:43,546 - INFO #  ===== Test begins =====
2016-03-20 13:37:43,548 - INFO #  =======================
2016-03-20 13:37:43,550 - INFO # Starting thread 3
2016-03-20 13:37:43,551 - INFO # sending 1st msg
2016-03-20 13:37:43,552 - INFO # sending 2nd msg
2016-03-20 13:37:43,553 - INFO # Starting thread 4
2016-03-20 13:37:43,554 - INFO # sending 3rd msg
2016-03-20 13:37:43,559 - INFO # sending 4th msg
2016-03-20 13:37:43,560 - INFO # Starting thread 5
2016-03-20 13:37:43,578 - INFO # now=0.094593 (0 hours 0 min), diff=0
2016-03-20 13:37:44,577 - INFO # now=1.094593 (0 hours 0 min), diff=0
2016-03-20 13:37:45,577 - INFO # now=2.094593 (0 hours 0 min), diff=0
2016-03-20 13:37:46,576 - INFO # now=3.094593 (0 hours 0 min), diff=0
2016-03-20 13:37:47,576 - INFO # now=4.094604 (0 hours 0 min), diff=11
2016-03-20 13:37:48,575 - INFO # now=5.094593 (0 hours 0 min), diff=0
2016-03-20 13:37:49,576 - INFO # now=6.094593 (0 hours 0 min), diff=0
2016-03-20 13:37:50,576 - INFO # now=7.094604 (0 hours 0 min), diff=11
2016-03-20 13:37:51,575 - INFO # now=8.094593 (0 hours 0 min), diff=0
2016-03-20 13:37:52,575 - INFO # now=9.094593 (0 hours 0 min), diff=0
2016-03-20 13:37:53,575 - INFO # now=10.094593 (0 hours 0 min), diff=0
2016-03-20 13:37:54,575 - INFO # now=11.094593 (0 hours 0 min), diff=0
2016-03-20 13:37:55,575 - INFO # now=12.094593 (0 hours 0 min), diff=0
2016-03-20 13:37:56,575 - INFO # now=13.094593 (0 hours 0 min), diff=0
2016-03-20 13:37:57,575 - INFO # now=14.094593 (0 hours 0 min), diff=0
2016-03-20 13:37:58,575 - INFO # now=15.094593 (0 hours 0 min), diff=0
2016-03-20 13:37:59,575 - INFO # now=16.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:00,575 - INFO # now=17.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:01,575 - INFO # now=18.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:02,575 - INFO # now=19.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:03,575 - INFO # now=20.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:04,575 - INFO # now=21.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:05,576 - INFO # now=22.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:06,576 - INFO # now=23.094646 (0 hours 0 min), diff=53
2016-03-20 13:38:07,576 - INFO # now=24.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:08,575 - INFO # now=25.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:09,575 - INFO # now=26.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:10,576 - INFO # now=27.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:11,575 - INFO # now=28.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:12,576 - INFO # now=29.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:13,576 - INFO # now=30.094626 (0 hours 0 min), diff=33
2016-03-20 13:38:14,576 - INFO # now=31.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:15,575 - INFO # now=32.094608 (0 hours 0 min), diff=15
2016-03-20 13:38:16,575 - INFO # now=33.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:17,575 - INFO # now=34.094611 (0 hours 0 min), diff=18
2016-03-20 13:38:18,576 - INFO # now=35.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:19,575 - INFO # now=36.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:20,575 - INFO # now=37.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:21,575 - INFO # now=38.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:22,576 - INFO # now=39.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:23,575 - INFO # now=40.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:24,575 - INFO # now=41.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:25,576 - INFO # now=42.094608 (0 hours 0 min), diff=15
2016-03-20 13:38:26,574 - INFO # now=43.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:27,575 - INFO # now=44.094636 (0 hours 0 min), diff=43
2016-03-20 13:38:28,575 - INFO # now=45.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:29,576 - INFO # now=46.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:30,576 - INFO # now=47.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:31,575 - INFO # now=48.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:32,576 - INFO # now=49.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:33,576 - INFO # now=50.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:34,576 - INFO # now=51.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:35,576 - INFO # now=52.094608 (0 hours 0 min), diff=15
2016-03-20 13:38:36,575 - INFO # now=53.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:37,576 - INFO # now=54.094617 (0 hours 0 min), diff=24
2016-03-20 13:38:38,575 - INFO # now=55.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:39,574 - INFO # now=56.094643 (0 hours 0 min), diff=50
2016-03-20 13:38:40,574 - INFO # now=57.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:41,574 - INFO # now=58.094608 (0 hours 0 min), diff=15
2016-03-20 13:38:42,574 - INFO # now=59.094593 (0 hours 0 min), diff=0
2016-03-20 13:38:43,576 - INFO # now=60.094637 (0 hours 1 min), diff=44
2016-03-20 13:38:44,574 - INFO # now=61.094593 (0 hours 1 min), diff=0
2016-03-20 13:38:45,581 - INFO # now=62.094593 (0 hours 1 min), diff=0
2016-03-20 13:38:46,574 - INFO # now=63.094608 (0 hours 1 min), diff=15
2016-03-20 13:38:47,581 - INFO # now=64.094593 (0 hours 1 min), diff=0
2016-03-20 13:38:48,574 - INFO # now=65.094593 (0 hours 1 min), diff=0
2016-03-20 13:38:49,579 - INFO # now=66.094593 (0 hours 1 min), diff=0
2016-03-20 13:38:50,574 - INFO # now=67.094593 (0 hours 1 min), diff=0
2016-03-20 13:38:51,574 - INFO # now=68.094644 (0 hours 1 min), diff=51
2016-03-20 13:38:52,574 - INFO # now=69.094593 (0 hours 1 min), diff=0
2016-03-20 13:38:53,574 - INFO # now=70.094593 (0 hours 1 min), diff=0
2016-03-20 13:38:54,576 - INFO # now=71.094608 (0 hours 1 min), diff=15
```

PeterKietzmann, 2016-03-21:
@srisler I don't understand your problem. Is it fixed or unrelated to this issue or why did you close it?

srisler, 2016-03-21:
Here is how it went: I had a code (using a nrf24l01) with hwtimer and
vtimer that worked. Then I updated Riot base that introduced xtimer. The
Nrf does not work with up to date code  (no message received by the
receiver).

As the (supposed) only difference was timer management I tried to change
some timing constant and doubling the DELAY..._US fixed the problem. Then I
shared the info.

But with time I realized that Nrf was very unstable (for unknown reason so
far). Then I think doubling the constant has nothing to do with my problem.
Today, I still can't make the nrf work and I'm still investigating.

Is it more clear to you?

Have a good day

PeterKietzmann, 2016-03-21:
A bit. Could you also share some information about when exactly your problem occurs? E.g., does it occur when running *tests/nrf24l01p_lowlevel*? I will reopen that issue an test it with an stm32f4discovery board. Probably at the next Hack'n'ACK on 29th march.

srisler, 2016-03-24:
Hi Peter,

Sorry I haven't seen your comment. I haven't run any the nrf24l01 test.

Actually I have a lot of problem with stm32f3discovery and I don't know why:
* Message going from one thread to another randomly caused the receiver to be "locked":
```
016-03-24 22:36:33,494 - INFO # msg_send() core/msg.c:97: Sending from 4 to 3. block=1 src->state=6 target->state=7
2016-03-24 22:36:33,496 - INFO # msg_send() core/msg.c:101: Target 3 is not RECEIVE_BLOCKED.
2016-03-24 22:36:33,505 - INFO # queue_msg(): queuing message
2016-03-24 22:36:33,508 - INFO # msg_send() core/msg.c:106: Target 3 has a msg_queue. Queueing message.
2016-03-24 22:36:33,508 - INFO # Asap_Prot: Message sent to upper layer
2016-03-24 22:36:33,513 - INFO # _msg_receive: 4: _msg_receive(): No thread in waiting list.
2016-03-24 22:36:33,517 - INFO # _msg_receive(): 4: No msg in queue. Going blocked.
2016-03-24 22:36:33,522 - INFO # _msg_receive: 3: _msg_receive(): We've got a queued message.
2016-03-24 22:36:33,531 - INFO # _msg_receive: 3: _msg_receive(): No thread in waiting list.
---------- NORMAL BEHAVIOUR ----------
--------- AFTER A WHILE ----------
2016-03-24 22:36:45,499 - INFO # msg_send() core/msg.c:97: Sending from 4 to 3. block=1 src->state=6 target->state=7
2016-03-24 22:36:45,505 - INFO # msg_send() core/msg.c:101: Target 3 is not RECEIVE_BLOCKED.
2016-03-24 22:36:45,513 - INFO # queue_msg(): queuing message
2016-03-24 22:36:45,514 - INFO # msg_send() core/msg.c:106: Target 3 has a msg_queue. Queueing message.
2016-03-24 22:36:45,518 - INFO # Asap_Prot: Message sent to upper layer
2016-03-24 22:36:45,524 - INFO # _msg_receive: 4: _msg_receive(): No thread in waiting list.
2016-03-24 22:36:45,528 - INFO # _msg_receive(): 4: No msg in queue. Going blocked.
-------- HERE THREAD 3 IS LOCKED --------
```

* When 6 threads are running, including 1 controlling an ESP8266), the Nrf24l01 stops receiving message (while there is not problem when 4 threads are running)
* Different behaviour when enabling DEBUG on some files...

Well, this is very unstable and I don't know why.

I can't really explain why, but it seems to be worse than with hw and vtimer. At least my system worked for a little while (but crashed at some point anyway).

I keep trying to debug it one by one

OlegHahm, 2016-03-25:
Can you check why the thread is blocked, i.e. check its status?

PeterKietzmann, 2016-03-29:
@srisler I *assume* your problems are not driver related but more to the stm32f3. However, I don't have that board available but I could still check the nrf24l01+ driver with an stm32f4 and **tests/nrf24l01p_lowlevel**

PeterKietzmann, 2016-03-30:
@srisler I tried *tests/nrf24l01p_lowlevel* between two transceivers, one connected to stm32f0discovery and one connected to stm32f4discovery. Except the fact that I needed to change the IRQ pin in the Makefile of the application (on stm32f4discovery PA2 is used for UART by default), everything worked as expected. I was able to initialize the devices and transfer data from one to the other and vice versa. I will let them run for a while and see what happens...

basilfx, 2016-03-30:
FWIW: I am also busy with the nRF24L01+ modules and I had to add ~10uF capacitor across power supply to improve stability because it did could not deliver enough power when transmitting. Although I don't use an STM32, it's something you can try.

PeterKietzmann, 2016-03-30:
The test applications ran about 2h and still work. But still, I will let this issue open. Please report your latest outcome and close the issue **if** you can confirm it's not the driver causing problems.

srisler, 2016-04-07:
Hi all,

After endless investigation I finally found a way to make it work. But it is not a very "clean" solution: It depends if I enable or not the "DEBUG trace" in a particular thread (which is not the one the nrf is driven from...). Hence, it has nothing to do with the driver. However DEBUG feature should not change anything to the program behaviour right?

Should I close this issue and create another one to talk about it?

OlegHahm, 2016-04-07:
> However DEBUG feature should not change anything to the program behaviour right?

Well, using DEBUG adds printing over the UART, which rather heavily impacts the timings of your application.

srisler, 2016-04-08:
Yes. "Heavily" is a bit surprising... Anyway, I close this issue and will take extra care when using debug.

Thank you for your help,

closed 2016-04-08 by srisler