RIOT-OS/RIOT #5061: gnrc_rpl: Sends multiple (identical?) DIO messages

gebart, 2016-03-14:
When I initialize the gnrc_networking example as an RPL root node the node starts sending out multiple similar or identical DIO messages in quick succession.

I don't know why this happens, is there a wrong default parameter which sets the trickle timer values to very short timeouts or is there a bug in the RPL DIO handling? @cgundogan maybe knows more?

Cloudshark link:
https://www.cloudshark.org/captures/5331a75b9497
pcap file:
https://www.dropbox.com/s/udbuff6qprsivbx/rpl-dio-spam2.pcapng?dl=0

packets 1-12 are from the RPL root node before starting the other nodes.
at t=29 s a second node runs rpl init
packets 13-36 are the transactions between the other gnrc_networking node and the root node for joining the DODAG, it should not need to be more than DIS->DIO->DAO->DAO_ACK->DIO, but in this dump the DIO messages are repeated many times over and over.

OlegHahm, 2016-03-14:
Isn't that the same as described here: https://github.com/RIOT-OS/RIOT/issues/4234

gebart, 2016-03-14:
@OlegHahm Seems like it could be, yes.

Now how about making this configurable somehow? 8 ms is an impossibly short time in a 6lowpan scenario. It will barely let anyone else use the channel for anything else until the RPL timer has gone up.

OlegHahm, 2016-03-14:
> Now how about making this configurable somehow?

Sounds like a wise idea. @cgundogan, @BytesGalore, anything's speaking against that?

BytesGalore, 2016-03-14:
@gebart @OlegHahm no objection to prevent DIO-storms from my side.
I guess it would be preferable to make it configurable on compile time and keeping the default value of 3^2 ms.

cgundogan, 2016-03-22:
@gebart can we close this now that #5073 was merged?

cgundogan, 2016-03-23:
fixed by #5073 and clarified in #4234

closed 2016-03-23 by cgundogan