RIOT-OS/RIOT #5062: shell ifconfig: Addresses added with ifconfig add should have lifetime infinite

gebart, 2016-03-14:
Currently the valid and preferred lifetimes of addresses added via the ifconfig shell command gets set to zero. This can be seen in packet dumps from a manually configured RPL root node, e.g. the packet dump in #5061 
See packet number 3 in https://www.cloudshark.org/captures/5331a75b9497

Somehow the RPL child nodes still manage to parse this into a lifetime of infinite, which may be another bug (or maybe an intentional fallback?)

gebart, 2016-03-16:
@authmillenon ping with relation to #4304

gebart, 2016-03-16:
the manual test in the cloudshark dump was run from gnrc_networking example, without USEMODULE border router.

cgundogan, 2016-03-23:
@gebart can this be closed as #4304 was merged a week ago?

gebart, 2016-03-23:
Yes

closed 2016-03-23 by gebart