RIOT-OS/RIOT #5066: objdump target broken for some boards

LudwigKnuepfer, 2016-03-14:
Compare https://github.com/RIOT-OS/RIOT/pull/4887#issuecomment-196537220

BytesGalore, 2016-03-15:
confirmed.
The `OBJDUMP` variable in the `Makefile.include` seems to be empty when called in the mentioned cases. You can reproduce it by setting an `@echo` before the call [1].
Strange that the test above the call doesn't fire.

Setting `OBJDUMP` manually works as expected (e.g. in `RIOT/tests/unittests`):
`make OBJDUMP=arm-none-eabi-objdump BOARD=msba2 tests-fib`

[1] https://github.com/RIOT-OS/RIOT/blob/master/Makefile.include#L333

OlegHahm, 2016-03-15:
I don't even understand why it is still working for Cortex-M.

hexluthor, 2016-03-15:
It works on Cortex-M because `Makefile.include.cortexm_common` pulls-in `Makefile.include.gnu`. I mistakenly assumed that all gnu targets would grab `Makefile.include.gnu`, but that is apparently not the case. My tentative plan is to just set `OBJDUMP` everywhere `OBJCOPY` is set. I wonder if it makes sense to have a makefile that adds default values for any variables the BOARD fails to set, `OBJDUMP ?= $(PREFIX)objdump` being an obvious choice.

OlegHahm, 2016-03-15:
Ah, yes, sorry, I figured that out after a while. My problem was that I was grepping for `Makefile.include.gnu`, but `gnu` was actually hidden in the `TOOLCHAIN` variable. I came up with #5077 then.

PR #5077 (2016-03-16): make: always include toolchain dependent Makefile
fixes #5066
closed 2016-03-16 by authmillenon