RIOT-OS/RIOT #5083: AVR8: Reboots after `return 0;`

kYc0o, 2016-03-15:
While testing `tests/xtimer_usleep_until` I noticed that the Arduino MEGA does not print the tests results, but only `Min/max error: xx/xxx`.
If the UART baud rate is increased to 115200 the complete test is printed, but then the platform reboot and this infinitely.

I'm porting another device ( [Waspmote PRO v1.2](http://www.libelium.com/products/waspmote/hardware/) ) which is based on rather the same MCU and I'm also experiencing this behaviour but at any baud rate.

Any hits to solve this mystery? 

Thanks RIOTers!

Paco.

kYc0o, 2016-07-26:
Still an issue. Postponed for the next release.

kYc0o, 2016-11-07:
This test has changed by `xtimer_periodic_wakeup` and now it succeeds. I think it was solved by #5763 .

closed 2016-11-07 by kYc0o

miri64, 2016-11-08:
At this stage of the release cycle: Please, please, please remove the issue from the "Release 2016.10" project and the release note drafts!

kYc0o, 2016-11-08:
Ok ok ok... Sorry, will do it asap.

miri64, 2016-11-08:
Already done. ;-) Just a reminder for next time