RIOT-OS/RIOT #5092: Build-time configuration header

gebart, 2016-03-16:
The command lines for the compilation are getting quite long with all the modules listed on the command line.
An alternate solution would be to introduce a build-time generated header that will contain defines for all enabled modules and other defines currently given on the command line. The drawback is that this configuration header must be included by all headers and c files in the tree..

something like `$(BINDIR)/riot-build.h`:
```c
#define COREIF_NG 1
#define CPU_ARCH_CORTEX_M4
#define CPU_MODEL_K60DN512VLL10
#define RIOT_BOARD BOARD_MULLE
...
#define MODULE_AT86RF212B
#define MODULE_AT86RF2XX
#define MODULE_GNRC_NETAPI
...
```

gebart, 2016-03-16:
Sample command line from the current build system to show how bad it is:
`arm-none-eabi-gcc \
	-DRIOT_FILE_RELATIVE=\"core/msg.c\" \
	-DRIOT_FILE_NOPATH=\"msg.c\" \
	-DMICROCOAP_DEBUG -Werror -DFEATURE_PERIPH_RTT -DFEATURE_PERIPH_SPI -DFEATURE_PERIPH_TIMER -DMODULE_AT86RF212B -DMODULE_AT86RF2XX -DMODULE_AUTO_INIT -DMODULE_AUTO_INIT_GNRC_NETIF -DMODULE_BITFIELD -DMODULE_BOARD -DMODULE_CONN -DMODULE_CONN_UDP -DMODULE_CORE -DMODULE_CORTEXM_COMMON -DMODULE_CPU -DMODULE_FIB -DMODULE_GNRC -DMODULE_GNRC_CONN -DMODULE_GNRC_CONN_UDP -DMODULE_GNRC_ICMPV6 -DMODULE_GNRC_ICMPV6_ECHO -DMODULE_GNRC_IPV6 -DMODULE_GNRC_IPV6_HDR -DMODULE_GNRC_IPV6_NC -DMODULE_GNRC_IPV6_NETIF -DMODULE_GNRC_IPV6_ROUTER -DMODULE_GNRC_IPV6_ROUTER_DEFAULT -DMODULE_GNRC_NDP -DMODULE_GNRC_NDP_INTERNAL -DMODULE_GNRC_NETAPI -DMODULE_GNRC_NETIF -DMODULE_GNRC_NETIF_DEFAULT -DMODULE_GNRC_NETIF_HDR -DMODULE_GNRC_NETREG -DMODULE_GNRC_NOMAC -DMODULE_GNRC_PKT -DMODULE_GNRC_PKTBUF -DMODULE_GNRC_PKTBUF_STATIC -DMODULE_GNRC_RPL -DMODULE_GNRC_SIXLOWPAN -DMODULE_GNRC_SIXLOWPAN_CTX -DMODULE_GNRC_SIXLOWPAN_FRAG -DMODULE_GNRC_SIXLOWPAN_IPHC -DMODULE_GNRC_SIXLOWPAN_ND -DMODULE_GNRC_SIXLOWPAN_ND_ROUTER -DMODULE_GNRC_SIXLOWPAN_NETIF -DMODULE_GNRC_SIXLOWPAN_ROUTER -DMODULE_GNRC_SIXLOWPAN_ROUTER_DEFAULT -DMODULE_GNRC_UDP -DMODULE_IEEE802154 -DMODULE_INET_CSUM -DMODULE_IPV6_ADDR -DMODULE_IPV6_HDR -DMODULE_KINETIS_COMMON -DMODULE_KINETIS_COMMON_PERIPH -DMODULE_MICROCOAP -DMODULE_NETIF -DMODULE_NEWLIB -DMODULE_NEWLIB_SYSCALLS_DEFAULT -DMODULE_NVRAM_SPI -DMODULE_PERIPH -DMODULE_PRNG_TINYMT32 -DMODULE_RANDOM -DMODULE_SHELL_COMMANDS -DMODULE_SIXLOWPAN -DMODULE_SYS -DMODULE_TINYMT32 -DMODULE_TRICKLE -DMODULE_TSRB -DMODULE_UART_STDIO -DMODULE_UDP -DMODULE_UNIVERSAL_ADDRESS -DMODULE_XTIMER -DCPU_MODEL_K60DN512VLL10 -DCPU_ARCH_CORTEX_M4 -mcpu=cortex-m4 -mlittle-endian -mthumb -mfloat-abi=soft -mno-thumb-interwork -ffunction-sections -fdata-sections -fno-builtin -fshort-enums -ggdb -g3 -Os -DCOREIF_NG=1 -DCPU_MODEL_K60DN512VLL10 -DCPU_ARCH_CORTEX_M4 -DBOARD_MULLE=\"mulle\" -DRIOT_BOARD=BOARD_MULLE -DCPU_K60=\"k60\" -DRIOT_CPU=CPU_K60 -DMCU_K60=\"k60\" -DRIOT_MCU=MCU_K60 -std=gnu99 -fno-delete-null-pointer-checks -Wstrict-prototypes -Werror=strict-prototypes -Wold-style-definition -Werror=old-style-definition -fno-common -Wall -DNDEBUG -DRIOT_VERSION=\"2016.03-devel-1194-g7dc661-c41aacac26bf-pr/microcoap-strlen\"  -isystem /usr/arm-none-eabi/include -I/data/riotbuild/riotbase/core/include -I/data/riotbuild/riotbase/drivers/include -I/data/riotbuild/riotbase/sys/include -I/data/riotbuild/riotcpu/k60/include -I/data/riotbuild/riotboard/mulle/include -I/data/riotbuild/riotcpu/kinetis_common/include -I/data/riotbuild/riotcpu/cortexm_common/include -I/data/riotbuild/riotbase/sys/libc/include -I/data/riotbuild/riotbase/sys/posix/include -I/data/riotbuild/riotbase/sys/net/include -I/data/riotbuild/riotbase/drivers/at86rf2xx/include -I/data/riotbuild/riotbase/pkg/microcoap/microcoap -MD -MP -c -o /data/riotbuild/riotproject/examples/microcoap_server/bin/mulle/core/msg.o /data/riotbuild/riotbase/core/msg.c`

gebart, 2016-03-16:
After introducing a header:

`arm-none-eabi-gcc \
	-DRIOT_FILE_RELATIVE=\"core/msg.c\" \
	-DRIOT_FILE_NOPATH=\"msg.c\" \
	-include /data/riotbuild/riotproject/examples/microcoap_server/bin/mulle/riot-build.h -Werror -DMICROCOAP_DEBUG -mcpu=cortex-m4 -mlittle-endian -mthumb -mfloat-abi=soft -mno-thumb-interwork -ffunction-sections -fdata-sections -fno-builtin -fshort-enums -ggdb -g3 -Os -std=gnu99 -fno-delete-null-pointer-checks -Wstrict-prototypes -Werror=strict-prototypes -Wold-style-definition -Werror=old-style-definition -fno-common -Wall -isystem /usr/arm-none-eabi/include -I/data/riotbuild/riotbase/core/include -I/data/riotbuild/riotbase/drivers/include -I/data/riotbuild/riotbase/sys/include -I/data/riotbuild/riotcpu/k60/include -I/data/riotbuild/riotboard/mulle/include -I/data/riotbuild/riotcpu/kinetis_common/include -I/data/riotbuild/riotcpu/cortexm_common/include -I/data/riotbuild/riotbase/sys/libc/include -I/data/riotbuild/riotbase/sys/posix/include -I/data/riotbuild/riotbase/sys/net/include -I/data/riotbuild/riotbase/drivers/at86rf2xx/include -I/data/riotbuild/riotbase/pkg/microcoap/microcoap -MD -MP -c -o /data/riotbuild/riotproject/examples/microcoap_server/bin/mulle/core/msg.o /data/riotbuild/riotbase/core/msg.c`

authmillenon, 2016-03-16:
TBH I find the output quite useful (if it is too verbose you can always deactivate it with `QUIET=1`). I would however like to have the output a little bit more structured (sorting by type of parameters, line-breaks to make it easily grep-able etc.)

authmillenon, 2016-03-16:
I would however really be happy about some header for stuff like DEVELHELP, etc.

gebart, 2016-03-16:
I think it would make more sense to put the definitions in a separate file since then you can look at them after building even if you closed your terminal window.

gebart, 2016-03-16:
the most annoying thing with the current situation is when you want to copy the command and modify something for an experiment when debugging since you have to scan through piles of defines to find the right place to edit.

Kijewski, 2016-03-16:
I like :+1:. There is little information gain in reading "… -DMODULE_GNRC_NDP_INTERNAL -DMODULE_GNRC_NETAPI -DMODULE_GNRC_NETIF …" over and over again.

kaspar030, 2016-03-16:
I've been writing a new build system, which is *about* ready. It tackles this with a different approach: every module specifies not only it's dependencies, but also which modules would modify it.
In the build files, it looks like this:

```
Module("gnrc_ipv6").needs(["gnrc_ipv6_addr", "gnrc_blah"]).uses("gnrc_sixlowpan")
```

The build system then dynamically creates the list of needed ```-DMODULE_...``` arguments.
The list become *much* shorter, IMHO mitigating the problem, with the advantage that files compiled for different applications can be cached more easily.

kaspar030, 2016-03-16:
... but, as long as we go the "-include build_data.h" route, it is definately an improvement.

gebart, 2016-03-16:
OT: @kaspar030 will your new build system require python for building or does it generate regular makefiles?

gebart, 2016-03-16:
I'll prepare a PR to show the concept.

gebart, 2016-03-17:
See #5097 for my proposal

authmillenon, 2016-03-17:
> OT: @kaspar030 will your new build system require python for building or does it generate regular makefiles?

I'm wondering: is that relevant? Both need to be installed on most systems (or are already installed).

PR #5097 (2016-07-07): make: Place most configuration macros in a separate file instead of on the command line
Fixes #5092 
Additionally fixes the known issue that a change to USEMODULE in the Makefile requires a manual `make clean` to build correctly (I couldn't find a Github issue for this though). Whenever USEMODULE (or CFLAGS) is modified in the Makefile, then all object files will be rebuilt.

The implementation checks the old file against the new file and only replaces the old file if the new file is different, this allows make to handle dependencies as usual by looking at the file timestamps.
closed 2016-07-07 by gebart