RIOT-OS/RIOT #5102: hashes: be32enc_vect in sha256.c should support unaligned memory access

wentaoshang, 2016-03-18:
`be32enc_vect`defined in sha256.c should support unaligned memory access, like in memcpy. Right now it assumes the input pointers are aligned over 4-byte boundaries.

Example: a function allocates memory of X bytes, computes the SHA-256 digest/HMAC of the first (X - 32) bytes, and copies the result to the last 32 bytes of that allocated memory block.

The example above will cause hard fault on SAM R21 board.

haukepetersen, 2016-04-20:
will not be addressed in this release, postponed

kYc0o, 2016-07-11:
anyone looking into this issue?

Kijewski, 2016-07-11:
Fixing this would be easy enough:

```c
static void be32enc_vect(void *dst_, const void *src_, size_t len)
{
    uint32_t *dst = dst_;   // we know that the destination is aligned
    const char *src = src_; // only the source is user supplied

    for (size_t i = 0; i < len / 4; i++) {
        dst[i] = ((uint32_t) src[4*i + 0] <<  0) |
                 ((uint32_t) src[4*i + 1] <<  8) |
                 ((uint32_t) src[4*i + 2] << 16) |
                 ((uint32_t) src[4*i + 3] << 24);
    }
}
```

(Untested, order of bytes could be wrong.)

wentaoshang, 2016-07-11:
We don't know for sure that destination will be aligned. The hash value
could be filled into any location in memory.

wentaoshang, 2016-07-11:
See my temporary fix for this issue. This is essentially the same code as memcpy.

https://github.com/wentaoshang/RIOT/commit/11c9baab117a56fa228a2c2c083be365f9bf3fa4

BytesGalore, 2016-07-26:
@wentaoshang you're right, the user passes destination and source to the functions which pass them further to the `be32enc_vect(..)`function.
Since `dst_` is a `void*` we cannot make any assumptions on its alignment.
I guess your temporary fix would do, but I wonder if we would only require the ``` else ``` part.

kYc0o, 2016-07-26:
Can anyone point to the PR solving this (if there is one)?

BytesGalore, 2016-07-26:
@kYc0o there is none currently. 
@wentaoshang could you provide a PR for the changes you made in https://github.com/wentaoshang/RIOT/commit/11c9baab117a56fa228a2c2c083be365f9bf3fa4, then we can discuss your proposition there and probably merge it soon :)

kYc0o, 2016-07-26:
Ok right, thanks. Let's try to do it for the next release.

wentaoshang, 2016-07-27:
Sorry I was distracted in the past few days. Will make a pull request ASAP.

OlegHahm, 2016-07-31:
@kYc0o, any reason not to integrate #5703 into the release?

kYc0o, 2016-07-31:
Nothing special, I didn't know that it was addressing this issue, I thought it was #5690, and the bug tag wasn't set in the PR. I'll look asap.

kYc0o, 2016-08-02:
Solved by #5703

closed 2016-08-02 by kYc0o