RIOT-OS/RIOT #5103: xtimer: weird behavior of tests/xtimer_drift, bug?

haukepetersen, 2016-03-18:
I am not sure if this is a bug in the `xtimer`, the `xtimer_drift` test but it is very likely. I got this broken behavior when running the `xtimer_drift` test on `nrf51` base boards (e.g. `airfy-beacon`, `nrf51dongle`):
```
2016-03-18 10:36:51,856 - INFO # now=15.397038 (0 hours 0 min), diff=0
2016-03-18 10:36:52,861 - INFO # now=16.397038 (0 hours 0 min), diff=0
2016-03-18 10:36:53,856 - INFO # now=17.397038 (0 hours 0 min), diff=0 // so far good
2016-03-18 10:37:28,411 - INFO # now=18.397038 (0 hours 0 min), diff=0 // jump of 35 sec
2016-03-18 10:37:46,190 - INFO # now=19.397038 (0 hours 0 min), diff=0 // jump of 18 sec
2016-03-18 10:37:47,190 - INFO # now=20.397038 (0 hours 0 min), diff=0
2016-03-18 10:37:48,190 - INFO # now=21.397038 (0 hours 0 min), diff=0
2016-03-18 10:37:49,190 - INFO # now=22.397038 (0 hours 0 min), diff=0
2016-03-18 10:37:50,191 - INFO # now=23.397038 (0 hours 0 min), diff=0
2016-03-18 10:38:07,966 - INFO # now=24.397038 (0 hours 0 min), diff=0 // jump of 17 sec
2016-03-18 10:38:08,966 - INFO # now=25.397038 (0 hours 0 min), diff=0
```
or
```
2016-03-18 10:49:04,860 - INFO # now=102.098898 (0 hours 1 min), diff=0
2016-03-18 10:49:05,860 - INFO # now=103.098898 (0 hours 1 min), diff=0
2016-03-18 10:49:23,637 - INFO # now=104.098898 (0 hours 1 min), diff=0 // jump
2016-03-18 10:49:24,637 - INFO # now=105.098898 (0 hours 1 min), diff=0
2016-03-18 10:49:25,637 - INFO # now=106.098898 (0 hours 1 min), diff=0
2016-03-18 10:49:26,637 - INFO # now=107.098898 (0 hours 1 min), diff=0
2016-03-18 10:49:27,637 - INFO # now=108.098898 (0 hours 1 min), diff=0
2016-03-18 10:49:28,637 - INFO # now=109.098898 (0 hours 1 min), diff=0
2016-03-18 10:49:29,637 - INFO # now=110.098898 (0 hours 1 min), diff=0
2016-03-18 10:49:47,414 - INFO # now=111.098898 (0 hours 1 min), diff=0 // jump
2016-03-18 10:49:48,414 - INFO # now=112.098898 (0 hours 1 min), diff=0
2016-03-18 10:49:49,414 - INFO # now=113.098898 (0 hours 1 min), diff=0
2016-03-18 10:49:50,414 - INFO # now=114.098898 (0 hours 1 min), diff=0
```
Those jumps appear on random intervals. 

I am ~90% convinced, that this is not caused by the peripheral timer driver, as the exact code is also running on the `nrf52dk`, where this behavior does not appear. It rather seems to me, that the xtimer (or the test app) has some concurrency issue, which shows only at the clock speed of the nrf51 boards (16MHz), but not at the higher speed of the nrf52 board.

I actually found this when debugging the timer for the `cc26x0` (#4675), where the test shows also very strange behavior...

haukepetersen, 2016-03-18:
Just out of interest, I have been running this test on the same CPUs:
```
#define TIM_SPEED           (1000000ul) /* try to run with 1MHz */

#define STEP                (10000U)
#define SUM                 (500)

static volatile int busy = 1;
static volatile int count = 0;

static void cb(void *arg, int chan)
{
    ++count;
    if (count == SUM) {
        count = 0;
        busy = 0;
    }

    timer_set(TIMER_DEV(0), 0, STEP);
}

int main(void)
{
    timer_init(TIMER_DEV(0), TIM_SPEED, cb, NULL);
    timer_set(TIMER_DEV(0), 0, STEP);

    while (1) {
        while (busy) {}
        busy = 1;
        puts("PING");
    }

    return 0;
}
```
With this, I get the `PING` message printed out reliable every 5s, for about half an hour now, without any pings skipped. So the issues seems only to appear when actually multiplexing timers?!

gebart, 2016-03-18:
I guess a missed deadline. Try increasing XTIMER_BACKOFF and XTIMER_OVERHEAD (or whatever the constants in xtimer.h are called, I'm too lazy to look them up right now)

haukepetersen, 2016-03-18:
already played with `XTIMER_SHIFT_ON_COMPARE`, same behavior for different values...

haukepetersen, 2016-03-18:
same for `XTIMER_BACKOFF`

gebart, 2016-03-18:
@haukepetersen is that the platform with 16 MHz timer fixed frequency?

what values did you test?

kaspar030, 2016-03-20:
@haukepetersen Is the underlying timer 32bit wide?

haukepetersen, 2016-03-21:
16-bit timer with a system clock of 16MHz

haukepetersen, 2016-03-21:
xtimer config is:
```c
#define XTIMER                      (0)
#define XTIMER_CHAN                 (0)
#define XTIMER_MASK                 (0xff000000)
#define XTIMER_SHIFT_ON_COMPARE     (2)
#define XTIMER_BACKOFF              (40)
```

haukepetersen, 2016-03-21:
and the (well functioning) `nrf52dk` runs at 48MHz with a 32-bit timer configuration.

gebart, 2016-03-21:
The mask is wrong

haukepetersen, 2016-03-21:
you mean it should be `0xffffff00`?

How about we spend some more documentation on the xtimer configuration values? And possibly put them into one easy to find block in the `xtimer.h`?

kaspar030, 2016-03-21:
> you mean it should be 0xffffff00?

No, he answered before you edited your post stating the timer was 16bit.

kaspar030, 2016-03-21:
> How about we spend some more documentation on the xtimer configuration values? And possibly put them into one easy to find block in the xtimer.h?

Not enough:
https://github.com/RIOT-OS/RIOT/blob/master/sys/include/xtimer.h#L369
?

haukepetersen, 2016-03-21:
actually it is enough. But it might be a good idea to move it (together with the other configuration options) to the top of the file, so it can easily be located?!

kaspar030, 2016-03-21:
Not the way we've been doing it...

http://doc.riot-os.org/group__sys__xtimer.html ?

kaspar030, 2016-03-21:
I'd love to help. How can I get one of the affected boards?

haukepetersen, 2016-03-21:
quote 'who ever looks at the f**ing doxygen' :-) But I guess you are right, that should be sufficient...

Back to this issue, so far I saw this behavior on the 'airfy-beacon' and 'nrf51dongle', and as stated not on the `nrf52dk`. Also I saw similar (or maybe different, but strange) behavior on the `cc2650stk` -> #4675

PeterKietzmann, 2017-01-26:
@haukepetersen @kaspar030 does this issue still persist???

haukepetersen, 2017-08-28:
Just checked, the issue at hand does still exist.

smlng, 2017-08-30:
cannot confirm behaviour on cc2650stk, I created a [gist](https://gist.github.com/smlng/9cdad410c899f7e55c1ee5723209f260) with the output of `tests/xtimer_drift` over the first 1000s. The only curious thing is that instead of having 1s between each output it falls short by roughly 6ms (994ms instead of 1000ms) between each print out. As the low level periph timer is running at 1MHz it cannot be some rounding offset but might rather be related to some of the internal offset when setting a timer.

kYc0o, 2018-07-17:
@Josar is this handled by one of your xtimer PRs?

Josar, 2018-07-17:
Why is ther no Test for this? So it can be run from command line?

Every line has errors in the log of a run on jiminy. 
Riot/master
```
2018-07-17 20:26:31,381 - INFO # now=1.044784 (0x0001fe26 ticks), drift=680 us, jitter=680 us
2018-07-17 20:26:33,955 - INFO # now=2.044368 (0x0003e63a ticks), drift=264 us, jitter=-416 us
2018-07-17 20:26:36,525 - INFO # now=3.043912 (0x0005ce49 ticks), drift=-192 us, jitter=-456 us
2018-07-17 20:26:38,569 - INFO # now=4.043848 (0x0007b689 ticks), drift=-256 us, jitter=-64 us
2018-07-17 20:26:40,618 - INFO # now=5.043784 (0x00099ec9 ticks), drift=-320 us, jitter=-64 us
2018-07-17 20:26:42,665 - INFO # now=6.043720 (0x000b8709 ticks), drift=-384 us, jitter=-64 us
2018-07-17 20:26:44,188 - INFO # now=7.044400 (0x000d6fa6 ticks), drift=296 us, jitter=680 us
2018-07-17 20:26:47,284 - INFO # now=8.043592 (0x000f5789 ticks), drift=-512 us, jitter=-808 us
2018-07-17 20:26:49,329 - INFO # now=9.044272 (0x00114026 ticks), drift=168 us, jitter=680 us
2018-07-17 20:26:51,901 - INFO # now=10.044000 (0x0013284c ticks), drift=-104 us, jitter=-272 us
2018-07-17 20:26:55,518 - INFO # now=11.043552 (0x0015105c ticks), drift=-552 us, jitter=-448 us
2018-07-17 20:26:57,565 - INFO # now=12.043336 (0x0016f889 ticks), drift=-768 us, jitter=-216 us
2018-07-17 20:27:01,705 - INFO # now=13.043272 (0x0018e0c9 ticks), drift=-832 us, jitter=-64 us
2018-07-17 20:27:04,801 - INFO # now=14.043632 (0x001ac93e ticks), drift=-472 us, jitter=360 us
2018-07-17 20:27:08,420 - INFO # now=15.043144 (0x001cb149 ticks), drift=-960 us, jitter=-488 us
2018-07-17 20:27:09,943 - INFO # now=16.043080 (0x001e9989 ticks), drift=-1024 us, jitter=-64 us
2018-07-17 20:27:13,036 - INFO # now=17.044224 (0x00208260 ticks), drift=120 us, jitter=1144 us
2018-07-17 20:27:14,036 - INFO # now=18.043728 (0x00226a6a ticks), drift=-376 us, jitter=-496 us
2018-07-17 20:27:16,082 - INFO # now=19.042888 (0x00245249 ticks), drift=-1216 us, jitter=-840 us
2018-07-17 20:27:20,224 - INFO # now=20.042824 (0x00263a89 ticks), drift=-1280 us, jitter=-64 us
2018-07-17 20:27:22,270 - INFO # now=21.042760 (0x002822c9 ticks), drift=-1344 us, jitter=-64 us
2018-07-17 20:27:25,364 - INFO # now=22.043504 (0x002a0b6e ticks), drift=-600 us, jitter=744 us
2018-07-17 20:27:27,935 - INFO # now=23.043256 (0x002bf397 ticks), drift=-848 us, jitter=-248 us
2018-07-17 20:27:29,457 - INFO # now=24.042568 (0x002ddb89 ticks), drift=-1536 us, jitter=-688 us
2018-07-17 20:27:31,505 - INFO # now=25.042504 (0x002fc3c9 ticks), drift=-1600 us, jitter=-64 us
2018-07-17 20:27:34,598 - INFO # now=26.042440 (0x0031ac09 ticks), drift=-1664 us, jitter=-64 us
2018-07-17 20:27:36,121 - INFO # now=27.042376 (0x00339449 ticks), drift=-1728 us, jitter=-64 us
2018-07-17 20:27:38,349 - INFO # now=28.042312 (0x00357c89 ticks), drift=-1792 us, jitter=-64 us
2018-07-17 20:27:41,445 - INFO # now=29.042712 (0x00376503 ticks), drift=-1392 us, jitter=400 us
2018-07-17 20:27:44,015 - INFO # now=30.042184 (0x00394d09 ticks), drift=-1920 us, jitter=-528 us
2018-07-17 20:27:46,585 - INFO # now=31.042120 (0x003b3549 ticks), drift=-1984 us, jitter=-64 us
2018-07-17 20:27:48,632 - INFO # now=32.042056 (0x003d1d89 ticks), drift=-2048 us, jitter=-64 us
2018-07-17 20:27:50,680 - INFO # now=33.041992 (0x003f05c9 ticks), drift=-2112 us, jitter=-64 us
2018-07-17 20:27:53,251 - INFO # now=34.041928 (0x0040ee09 ticks), drift=-2176 us, jitter=-64 us
```

Almost no error, one post every second, just a too fast running timer. But this is expected.
vs https://github.com/RIOT-OS/RIOT/pull/9211

```
2018-07-17 20:31:38,322 - INFO # now=1.046920 (0x0001ff31 ticks), drift=-64 us, jitter=-64 us
2018-07-17 20:31:39,321 - INFO # now=2.046856 (0x0003e771 ticks), drift=-128 us, jitter=-64 us
2018-07-17 20:31:40,320 - INFO # now=3.046792 (0x0005cfb1 ticks), drift=-192 us, jitter=-64 us
2018-07-17 20:31:41,318 - INFO # now=4.046728 (0x0007b7f1 ticks), drift=-256 us, jitter=-64 us
2018-07-17 20:31:42,317 - INFO # now=5.046664 (0x0009a031 ticks), drift=-320 us, jitter=-64 us
2018-07-17 20:31:43,315 - INFO # now=6.046600 (0x000b8871 ticks), drift=-384 us, jitter=-64 us
2018-07-17 20:31:44,316 - INFO # now=7.046688 (0x000d70c4 ticks), drift=-296 us, jitter=88 us
2018-07-17 20:31:45,315 - INFO # now=8.047280 (0x000f5956 ticks), drift=296 us, jitter=592 us
2018-07-17 20:31:46,315 - INFO # now=9.047208 (0x00114195 ticks), drift=224 us, jitter=-72 us
2018-07-17 20:31:47,313 - INFO # now=10.046344 (0x00132971 ticks), drift=-640 us, jitter=-864 us
2018-07-17 20:31:48,311 - INFO # now=11.046280 (0x001511b1 ticks), drift=-704 us, jitter=-64 us
2018-07-17 20:31:49,312 - INFO # now=12.046216 (0x0016f9f1 ticks), drift=-768 us, jitter=-64 us
2018-07-17 20:31:50,308 - INFO # now=13.046152 (0x0018e231 ticks), drift=-832 us, jitter=-64 us
2018-07-17 20:31:51,310 - INFO # now=14.046088 (0x001aca71 ticks), drift=-896 us, jitter=-64 us
2018-07-17 20:31:52,309 - INFO # now=15.046024 (0x001cb2b1 ticks), drift=-960 us, jitter=-64 us
2018-07-17 20:31:53,305 - INFO # now=16.046376 (0x001e9b25 ticks), drift=-608 us, jitter=352 us
2018-07-17 20:31:54,307 - INFO # now=17.045896 (0x00208331 ticks), drift=-1088 us, jitter=-480 us
2018-07-17 20:31:55,304 - INFO # now=18.045832 (0x00226b71 ticks), drift=-1152 us, jitter=-64 us
2018-07-17 20:31:56,304 - INFO # now=19.045768 (0x002453b1 ticks), drift=-1216 us, jitter=-64 us
2018-07-17 20:31:57,302 - INFO # now=20.045704 (0x00263bf1 ticks), drift=-1280 us, jitter=-64 us
[... cut ...]
2018-07-17 20:32:14,287 - INFO # now=37.044776 (0x0046a845 ticks), drift=-2208 us, jitter=96 us
2018-07-17 20:32:15,287 - INFO # now=38.044552 (0x00489071 ticks), drift=-2432 us, jitter=-224 us
2018-07-17 20:32:16,286 - INFO # now=39.044488 (0x004a78b1 ticks), drift=-2496 us, jitter=-64 us
2018-07-17 20:32:17,286 - INFO # now=40.045312 (0x004c6160 ticks), drift=-1672 us, jitter=824 us
2018-07-17 20:32:18,283 - INFO # now=41.044360 (0x004e4931 ticks), drift=-2624 us, jitter=-952 us
```

Maybe adding a test to this which checks if the timer is triggered every second, would be a good idea.

Josar, 2018-07-18:
@kYc0o the problem with the race condition is because of the high working load. This lead to a always to late target in `_xtimer_periodic_wakeup`. This can be fixed, but then it does not behave like described.

IMHO, its a bad idea to not update the `last_wakeup` if the set target is allready reached as this could lead to a allways late target depending on the period.

Josar, 2018-07-18:
@kYc0o please see https://github.com/RIOT-OS/RIOT/pull/9595 https://github.com/RIOT-OS/RIOT/pull/9596

kYc0o, 2018-07-18:
Thank you @Josar !

javierfileiv, 2019-03-14:
Thank you very much @Josar , I haven't fully tested it but I was having xtimer issues when using the kow power 16bit timer as xtimer source, and with your patch it seems to work! :)

javierfileiv, 2019-03-20:
Well, after some testing I can say that in the longterm test app, the 1 min printout is not present on the output. Sunno why is this happening.

MichelRottleuthner, 2020-01-24:
Was fixed with #9530 as can be seen from the testing results in the comments there.
I also checked with nrf51dk again and there are no more random delays/jumps in `tests/xtimer_drift`.

closed 2020-01-24 by MichelRottleuthner