RIOT-OS/RIOT #5108: hashes: hmac_sha256 should take input message of type "const unsigned char*"

wentaoshang, 2016-03-18:
`const unsigned char *hmac_sha256(const unsigned char *key,
                                 size_t key_length,
                                 const unsigned *message,
                                 size_t message_length,
                                 unsigned char *result)` defined in sys/hashes/sha256.c should take input "message" of type `const unsigned char*` or `const uint8_t*`, instead of `const unsigned*`, to be consistent with the `unsigned char *sha256(const unsigned char *d, size_t n, unsigned char *md)` API.

It is more common for the messages to be stored in unsigned char arrays. This suggested change would avoid extra type casting in many cases.

BytesGalore, 2016-07-26:
I proposed #5690

miri64, 2016-10-18:
I guess this was fixed (though differently than proposed) in #5690.

closed 2016-10-18 by miri64