RIOT-OS/RIOT #5128: make: buildtest breaks when exporting FEATURES_PROVIDED var

haukepetersen, 2016-03-22:
for #5065 (and possibly other improvements in the future), I exported the `FEATURES_PROVIDED` variable, so I can use it in some other Makefiles, e.g. in `Makefile.periph` to be the periph driver selectively.

But this breaks the `make buildtest`, as the `Makefile.buildtest` is doing some magic I don't understand...

So I can we fix this?

haukepetersen, 2016-03-29:
will not be solved for this release -> let's try the next

Kijewski, 2016-03-30:
[Makefile.buildtests#L41](https://github.com/RIOT-OS/RIOT/blob/3d9f66e91f07f3a4f6b415d29b37c7cfd0ac71c2/Makefile.buildtests#L41)/[#L254](https://github.com/RIOT-OS/RIOT/blob/3d9f66e91f07f3a4f6b415d29b37c7cfd0ac71c2/Makefile.buildtests#L254) are to blame. Either that or the person wrote wrote these lines. \*pfeif\*

Please try to add line `undefine FEATURES_PROVIDED` before both instances of `FEATURES_PROVIDED := $(FEATURES_PROVIDED_BAK)`. `undefined` only exists since GNU make 3.82 (2010-07-28), so there is a tiny chance that not all systems have this functionality.

kYc0o, 2016-07-26:
And postponed again...

kaspar030, 2016-08-17:
*bump*

> Please try to add line undefine FEATURES_PROVIDED before both instances of FEATURES_PROVIDED := $(FEATURES_PROVIDED_BAK). undefined only exists since GNU make 3.82 (2010-07-28), so there is a tiny chance that not all systems have this functionality.

doesn't help.

OlegHahm, 2017-01-15:
Postponing again?

haukepetersen, 2017-09-08:
fixed by #7507 and export will be added by #6690, so this issue is obsolete.

closed 2017-09-08 by haukepetersen