RIOT-OS/RIOT #5130: git-cache: problem when building for multiple platforms

OlegHahm, 2016-03-22:
I added the microcoap package to my local git-cache (as introduced in #5112). Now, if I want to build for multiple platforms, the second build fails. E.g. I build `examples/microcoap` for native (`make -B clean all`) and after that I try to build for samr21-xpro (`BOARD=samr21-xpro make -B clean all`). The second build fails with:
```
Cleaning up microcoap package...
fatal: Resolve operation not in progress, we are not resuming.
make: Nothing to be done for 'clean'.
Building application "microcoap_server" for "samr21-xpro" with MCU "samd21".

"make" -C /home/oleg/git/RIOT/pkg/microcoap
test -d "/home/oleg/git/RIOT/pkg/microcoap/microcoap" || /home/oleg/git/RIOT/dist/tools/git/git-cache clone "git://github.com/1248/microcoap.git" "9cb1dcda2182a8dca8483b230cda8b591a924c82" "/home/oleg/git/RIOT/pkg/microcoap/microcoap"
cd "/home/oleg/git/RIOT/pkg/microcoap/microcoap" && git am --ignore-whitespace "/home/oleg/git/RIOT/pkg/microcoap"/*.patch
Applying: Remove unneeded *.c files
error: endpoints.c: does not exist in index
error: main-posix.c: does not exist in index
error: microcoap.ino: does not exist in index
Patch failed at 0001 Remove unneeded *.c files
The copy of the patch that failed is found in: .git/rebase-apply/patch
When you have resolved this problem, run "git am --continue".
If you prefer to skip this patch, run "git am --skip" instead.
To restore the original branch and stop patching, run "git am --abort".
Makefile:19: recipe for target '/home/oleg/git/RIOT/pkg/microcoap/microcoap/Makefile' failed
make[1]: *** [/home/oleg/git/RIOT/pkg/microcoap/microcoap/Makefile] Error 128
/home/oleg/git/RIOT/examples/microcoap_server/../../Makefile.include:299: recipe for target '/home/oleg/git/RIOT/examples/microcoap_server/bin/samr21-xpro/microcoap.a' failed
make: *** [/home/oleg/git/RIOT/examples/microcoap_server/bin/samr21-xpro/microcoap.a] Error 2
```

*Update*
Since I get a similar error for #5126 I assume it's not related to a package being added to the local cache or not.

kaspar030, 2016-03-22:
It's the combination of "-B" and "clean" that breaks together with non-idempotent checking out and "clean" not actually cleaning the target.

Before, the "$(PKG_DIR)/.git/config" tried to clone, then whatever the return code would be, force-checkout a specific commit. Now, if the checkout directory already exists, it is left untouched. But "-B" forces another patch run, which then fails as the patches are already applied.

The workaround is not using "-B clean" together.
The quickfix would be to add another "checkout -f" to ""$(PKG_DIR)/.git/config" (dirty), or to make "clean" essentially do "distclean" (less dirty).

I'll make a PR.

authmillenon, 2016-03-22:
Is there still any need to use `clean` and `-B` together, I wonder... `clean` purges the entire build directory and applies `git clean -xf` on any package (effectively deleting any file not part of any commit) and checks-out a clean head. So `make clean all` should have the same effect as `make -B clean all` with the little difference that `make -B clean all` redownloads every package and patches it (which should not be required in most cases). The workaround that @kaspar030 is proposing should also be the case in general. Why `clean` when you force-execute every target anyway?

OlegHahm, 2016-03-22:
> Is there still any need to use clean and -B together

While I agree that it might not be necessary any more, it still shouldn't fail. Moreover, I'm very much used to call it this way just to be on the safe side. Nothing is more frustrating than hunting a but that  actually doesn't exist but is just a consequence of a broken build.

OlegHahm, 2016-03-22:
@kaspar030, thanks for the explanation.

closed 2016-03-23 by authmillenon

OlegHahm, 2016-03-23:
Confirmed that this has been fixed by #5138.

PR #5138 (2016-03-23): pkg: unify git checkout
This PR introduces a shared makefile for all the git checkout/patch/clean stuff.

Packages are now checked out into ```$(BINDIR)/pkg/<pkg_name>```, solving potential parallel-build-for-multiple-platform problems.
Patches went into ```patches``` subdirectories.

Fixes #5130 #4962 #4511.