RIOT-OS/RIOT #5132: SPI2 not working on stm32f1 cpu

aabadie, 2016-03-22:
Hi,
I started to port RIOT on the iotlab M3-A8 board but noticed an issue with SPI2. This interface is connected to the RF at86rf231 module (where it's using SPI1 on the iotlab-m3 board) and is the last problem remaining before having M3-A8 "fully" working on iotlab. The rest is working ok (l3g4200d, lsm303dlhc).

Digging into this SPI2 issue, I found that the spi pin configuration in [spi.c](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32f1/periph/spi.c#L161) was using gpio initialization functions where clocks are only for one port register (PA) (see [here](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32f1/periph/gpio.c#L88) and [here](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32f1/periph/gpio.c#L145).

Unfortunately, SPI2 is using register `APB1ENR` and there the clock are never enabled.

Adding, in the gpio init functions the line:
```RCC->APB1ENR |= (RCC_APB2ENR_IOPBEN << _port_num(pin));```
does the job but I know it's an ugly hack.

I pushed a [wip branch](https://github.com/aabadie/RIOT/tree/iotlab-a8-m3) with this port and the dirty fix in the cpu spi gpio. I tested it with success on an experiment on the iotlab.

I put @kYc0o and @adjih who might be interested in the discussion.

kYc0o, 2016-03-22:
@haukepetersen do you know why this register is written from there in this manner? I would maybe expect another parameter on this function to specify which register I want to modify (or find the register depending on the pin/port using this alternative feature).

PeterKietzmann, 2016-03-23:
Hi @aabadie. Not sure if I'm on the right track though but did you know:

- The transceiver driver uses GPIO initialization **and** SPI initialization functions
- In the [SPI driver](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32l1/periph/spi.c#L57) of the stm32f1 CPU the clock is initialized calling a macro 
- This macro is defined in the peripheral configuration file of the board. Compare for the iotlab-m3 node [here](https://github.com/RIOT-OS/RIOT/blob/master/boards/iotlab-m3/include/periph_conf.h#L125)
- You can also find a configuration script describing the boards configuration. It defines the port which the transceiver is connected to (compare [here](https://github.com/RIOT-OS/RIOT/blob/master/boards/iotlab-m3/include/board.h#L55))

So IMO if you (i) add an entry for SPI2 in your peripheral configuration file and (ii) configure the transceiver using that SPI port in the boards configuration file you should be fine. Please note that the SPI configuration lines might be adapted during the next weeks (compare  #4780)

aabadie, 2016-03-23:
Thanks for the reply @PeterKietzmann.

>    In the SPI driver of the stm32f1 CPU the clock is initialized calling a macro

Not sure to understand all the code in this macro.

 >   This macro is defined in the peripheral configuration file of the board. Compare for the iotlab-m3 node here

On a8-m3 board, the RF transceiver uses SPI2 on the CPU. I adapted this file accordingly [here](https://github.com/aabadie/RIOT/blob/iotlab-a8-m3/boards/iotlab-a8-m3/include/periph_conf.h#L125). I duplicated the code for simplicity (on my side).

 >   You can also find a configuration script describing the boards configuration. It defines the port which the transceiver is connected to (compare here)

I also made this change without success [here](https://github.com/aabadie/RIOT/blob/iotlab-a8-m3/boards/iotlab-a8-m3/include/board.h#L53). GPIO pins are not the same on the a8-m3.

But as I said on iotlab-m3, the transceiver uses SPI1 which itself is using register `APB2ENR`. On the iotlab-a8-m3, the tranceiver uses SPI2 which itself is using register `APB1ENR`. The thing is just to enable the right one in the initialization functions (spi or gpio).

PeterKietzmann, 2016-03-23:
@aabadie I'd propose to define a second SPI device (`SPI_1`) besides the already existing one to avoid confusion when comparing the m3 and the a8-m3 board. I don't see the change for GPIO pins in your code.
Probably I didn't get your problem but again, let me suggest how the SPI2 clock should get initialized:

- transceiver driver calls the SPI init function with the bus that is set in the board.h
- in the stm32f1 SPI driver a macro is called to enable the concerned SPI port
- this bus clock enabling macro is defined in the periph_conf.h
- the macro can set the bit in `APB1ENR`

Did I miss something -or- where do you get stuck?

kYc0o, 2016-03-23:
So maybe the issue is that there is a double initialisation that does not make sense: [here](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32f1/periph/spi.c#L162) and [here](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32f1/periph/spi.c#L63), which IMO makes [this](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32f1/periph/gpio.c#L145) and [this](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32f1/periph/gpio.c#L88) irrelevant. If not, what's the purpose of these both initialisations?

aabadie, 2016-03-23:
> I'd propose to define a second SPI device (SPI_1) besides the already existing one to avoid confusion when comparing the m3 and the a8-m3 board.

The difference between iotlab-m3 and iotlab-a8-m3 is that SPI1 is not used any more in the latter but replaced by SPI2. That's why I keep SPI_0 in RIOT but pointed to the M3 SPI2 interface.
See [hardware documentation](http://github.com/iot-lab/iot-lab/wiki/Docs/opena8-schematics.pdf)

> I don't see the change for GPIO pins in your code.

My bad, I pointed to the wrong place, those pins are the same. But the ones [here](https://github.com/aabadie/RIOT/blob/iotlab-a8-m3/boards/iotlab-a8-m3/include/periph_conf.h#L129) has to be changed (SCLK, MISO, MOSI).

About the initialization:
> * this bus clock enabling macro is defined in the periph_conf.h
> * the macro can set the bit in APB1ENR 
These 2 last points are IMO the problematic ones.

aabadie, 2016-03-23:
> So maybe the issue is that there is a double initialisation that does not make sense: here and here, which IMO makes this and this irrelevant. If not, what's the purpose of these both initialisations?

Maybe I'm wrong but those 2 initialization are not exactly the same:
*  in __periph_conf.h__:
```c
#define SPI_0_CLKEN()       (RCC->APB1ENR |= RCC_APB1ENR_SPI2EN)
#define SPI_0_CLKDIS()      (RCC->APB1ENR &= ~(RCC_APB1ENR_SPI2EN))
```
* in __gpio.c__:
```c
RCC->APB1ENR |= (RCC_APB2ENR_IOPBEN << _port_num(pin));
```

This works but I don't really understand why.

PeterKietzmann, 2016-03-23:
No they are not the same :-). With ``RCC->APB1ENR |= RCC_APB1ENR_SPI2EN` you want to set bit no 14 in the `APB2`**(two)**`ENR` register to enable SPI2. I don't know what you aim to do with `RCC->APB1ENR |= (RCC_APB2ENR_IOPBEN << _port_num(pin))` (even though the name of the register and the bit in the register don't match), but what you're actually doing is to set the `TIM7EN` bit in `APB1`*(one)*ENR`. I don't know that board (neither looked it up). Does it have some pinouts? It could be helpful to verify that SPI and GPIO work properly. Therefore you could use *test/periph_spi* and *tests/periph_gpio`

PeterKietzmann, 2016-03-23:
Oh wait I see some confusions between the reference manual and vendor header, one moment please :-)

PeterKietzmann, 2016-03-23:
Forget about the inconsistencies in the reference manual, my bad :-) ! The problem lies in the GPIO driver that does not allow to enable a clock in the `APB1ENR` register when initializing a pin as alternative function. See [here](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32f1/periph/gpio.c#L145). So you see it the same way? But still I don't understand why out code fixed it

aabadie, 2016-03-23:
> So you see it the same way? 

yes ;)

> But still I don't understand why out code fixed it

Me neither. I can start a new experiment and try `test/periph_spi` as you suggested.

PeterKietzmann, 2016-03-23:
Yes, please try to use SPI and GPIO separately. After thinking about it a bit more I don't think that's the problem. The SPI2 clock in `AP1ENR` is enabled in the SPI driver by calling the macro in the periph_conf. The SPI pin configuration calls functions of the GPIO driver, e.g. `gpio_init_af`. In there, the port clock for the IO is enabled which is in `APB2ENR` in any way. That seems quite ok to me.

aabadie, 2016-03-23:
@PeterKietzmann, @kYc0o, good news.
I started a new experiment on iotlab using a8-m3 boards, tried the periph_spi test application, with and without the non understandable change in gpio.c (`RCC->APB1ENR |= (RCC_APB2ENR_IOPBEN << _port_num(pin));`) : Both worked, so I guessed the change is useless and flashed the `default` example without it (only using the new definition in boards/iotlab-a8-m3).

The RF transceiver is working ! So I apologize if you lost time because of my fault.

I'll clean up the code and create a PR asap.

aabadie, 2016-03-23:
I guess the issue can be closed as well.

PeterKietzmann, 2016-03-23:
Happy to see it working, even if I couldn't really help...

closed 2016-03-23 by PeterKietzmann

kYc0o, 2016-03-23:
Great! Thanks @PeterKietzmann, you helped a lot clarifying the purpose of this functions ;)