RIOT-OS/RIOT #5145: Stack overflow in msg_send()

srisler, 2016-03-23:
Hi all,

I have a RIOT project running on a stm32f3discovery and after a while, it appears to get overflowed, apparently in the msg_send function:

```
2016-03-22 23:30:08,839 - INFO # Context before hardfault:
2016-03-22 23:30:08,841 - INFO #    r0: 0x2000165c
2016-03-22 23:30:08,841 - INFO #    r1: 0x000060e5
2016-03-22 23:30:08,847 - INFO #    r2: 0x00000001
2016-03-22 23:30:08,848 - INFO #    r3: 0x200000c8
2016-03-22 23:30:08,849 - INFO #   r12: 0x20000fa6
2016-03-22 23:30:08,850 - INFO #    lr: 0x08001d23
2016-03-22 23:30:08,850 - INFO #    pc: 0x080022fc
2016-03-22 23:30:08,854 - INFO #   psr: 0x01000000
2016-03-22 23:30:08,855 - INFO # 
2016-03-22 23:30:08,861 - INFO # FSR/FAR:
2016-03-22 23:30:08,862 - INFO #  CFSR: 0x00008200
2016-03-22 23:30:08,862 - INFO #  HFSR: 0x40000000
2016-03-22 23:30:08,864 - INFO #  DFSR: 0x00000008
2016-03-22 23:30:08,864 - INFO #  AFSR: 0x00000000
2016-03-22 23:30:08,866 - INFO #  BFAR: 0x2001845c
2016-03-22 23:30:08,866 - INFO # Misc
2016-03-22 23:30:08,867 - INFO # EXC_RET: 0xfffffffd
2016-03-22 23:30:08,867 - INFO # Attempting to reconstruct state for debugging...
2016-03-22 23:30:08,867 - INFO # In GDB:
2016-03-22 23:30:08,868 - INFO #   set $pc=0x80022fc
2016-03-22 23:30:08,870 - INFO #   frame 0
2016-03-22 23:30:08,870 - INFO #   bt
2016-03-22 23:30:08,870 - INFO # 
2016-03-22 23:30:08,874 - INFO # ISR stack overflowed by at least 8 bytes.
```

When make objdump:
```
static int _msg_send(msg_t *m, kernel_pid_t target_pid, bool block, unsigned state)
{
 80022f4:       e92d 41ff       stmdb   sp!, {r0, r1, r2, r3, r4, r5, r6, r7, r8, lr}
 80022f8:       461f            mov     r7, r3
    if (!pid_is_valid(target_pid)) {
        DEBUG("msg_send(): target_pid is invalid, continuing anyways\n");
    }
#endif /* DEVELHELP */

    thread_t *target = (thread_t*) sched_threads[target_pid];
 80022fa:       4b2b            ldr     r3, [pc, #172]  ; (80023a8 <_msg_send+0xb4>)
 80022fc:       f853 6021       ldr.w   r6, [r3, r1, lsl #2]

    m->sender_pid = sched_active_pid;
 8002300:       4b2a            ldr     r3, [pc, #168]  ; (80023ac <_msg_send+0xb8>)
 8002302:       881b            ldrh    r3, [r3, #0]
 8002304:       8003            strh    r3, [r0, #0]
    }
    return _msg_send(m, target_pid, false, disableIRQ());
}
```

It is coherent with the log, it happens when a message received on the radio side (based on NRF24L01) is sent to another thread for application processing. This stack overflow appears randomly but generally after hundreds of correct message reception. Stack for thread are correctly sized (checked with ps).

Any clew?

Thank you in advance

miri64, 2016-10-19:
Any information on your setup? From which thread to which did you send? Did you initialize the message queue properly (if you used them)?

kaspar030, 2016-11-18:
I'm inclined to close this. @srisler any objections?

srisler, 2016-11-18:
Yes you can. It nerver happened again when I made major change un code.

Thx

closed 2016-11-18 by srisler