RIOT-OS/RIOT #5147: CI: build with debug

OlegHahm, 2016-03-23:
Following the discussion in #5143, the question arose if the CI should additionally perform build tests with debug enabled.

Yonezawa-T2, 2016-03-23:
Options:

- enable debug only for the native platform
- enable debug only for the unittests
- enable debug only for the native platform and some other platforms
- enable debug all files for the native platform, only `cpu/*` and `board/*` for the other platforms.
- other?

kaspar030, 2016-03-23:
IMHO enabling debug for the native platform is definitely a good idea. It won't hurt at all, and displaying a backtrace is simple to implement and probably very useful.
Once we have the infrastructure for that in place, and we start running tests on actual hardware, we can try to get backtraces from real hardware.

Yonezawa-T2, 2016-03-24:
Can we expect compilers to eliminate `if (0) { ... }` block? GCC (arm-none-eabi-gcc) 5.1.0 and Clang 7.3.0 seem to eliminate dead codes at default optimization level.

kaspar030, 2016-03-24:
> Can we expect compilers to eliminate if (0) { ... } block?

Yes! Unfortunately clang still complains about tautotlogical compares (``` if (LOG_LEVEL<LOG_INFO) {...}```.

gebart, 2016-03-24:
As a workaround for the clang bug: cast the enum to unsigned int in the
comparison

kaspar030, 2016-03-24:
See #5166.

closed 2019-09-10 by stale[bot]