RIOT-OS/RIOT #5150: prng: how to seed the random number generator?

OlegHahm, 2016-03-23:
As @haukepetersen stated in today's meeting, this problem is two-fold:
1. How to multiplex the PRNG initialization for different sources for the source (and boards without a hardware PRNG)?
2. How to generate a source for randomness for boards without a hardware PRNG?

authmillenon, 2016-03-23:
I implemented something like that a while back and opened a PR for it now: #5153.

authmillenon, 2016-03-23:
(it offers a solution for both problem parameters).

authmillenon, 2016-03-23:
Fixed issue description.

authmillenon, 2016-03-23:
A value read with `DEV_RANDOM_READ` can then be used to initialize the random number generator in `auto_init` or somewhere like that.

kaspar030, 2016-03-23:
There's also #4294. It can use drift between two crystals when there are no "natural" random sources.

authmillenon, 2016-03-23:
We could include that into #5153. I still have a bad feeling about your spiking of the PNRG there.

kaspar030, 2016-03-23:
Spiking?

authmillenon, 2016-03-23:
The [introduction of entropy into the result chain of the PNRG](https://github.com/RIOT-OS/RIOT/pull/4294/files#diff-6bb28b9d35804f9328041bf90718a307R138) you did there.

gebart, 2016-04-30:
Some recent developments on Linux can be relevant here as well:

 - http://phoronix.com/scan.php?page=news_item&px=LRNG-dev-random-V2
 - http://www.chronox.de/lrng/doc/lrng.pdf
 - http://lkml.iu.edu/hypermail/linux/kernel/1604.3/00029.html

maribu, 2022-09-16:
@Einhornhool could maybe take a look at this. Out of my had I would say: Use `puf_sram` or a true hardware RNG.

Einhornhool, 2022-09-19:
For multiplexing different sources, the mbedTLS entropy module could be an option (#15671).
As a source of randomness on boards without a HWRNG, `puf_sram` can be used (like @maribu said).

maribu, 2022-09-19:
Looking in the code (`auto_init_random()` in `sys/random/random.c`), `puf_sram` or a hardware RNG are already used to seed the PRNG, when available. The fallback is to used `luid_get()` (which would at least result in different boards using different seeds) or a default constant.

I guess this means we should make try to get `puf_sram` supported for all boards and everything will be fine.