RIOT-OS/RIOT #5156: sensor drivers: redudant structs

OlegHahm, 2016-03-23:
The driver for, e.g. the lis3dh or l3g4200d contain redundant structs. `l3g4200d_t` and `lis3dh_t` are a subset of `l3g4200d_params_t` or `lis3dh_params_t`.

haukepetersen, 2016-03-29:
Good point, I think this is the good old issue of code duplication vs. the same concept for all device drivers. But I have the feeling, that for this case, we might want got get rid of this code duplication here. Will address this in terms of device driver cleanup once the auto_init concept is decided/solved (#4658).

kYc0o, 2016-07-26:
I hope this will be addressed on the next release :)

aabadie, 2018-04-16:
This issue is probably fixed by #8697 and #8698

PeterKietzmann, 2018-09-10:
@OlegHahm  #8697 and #8698 cleaned up the structures and have been merged so I'm closing this issue.

closed 2018-09-10 by PeterKietzmann