RIOT-OS/RIOT #5164: stm32 common i2c periph (?)

lebrush, 2016-03-24:
The implementation of the i2c periph is kind of similar in many stm32 platforms, so it might be worth to have one single implementation for all in `stm32_common`?

Maybe once the `DMA` is abstracted and can be used with API calls... (#5095)

DipSwitch, 2016-03-24:
The I2C is the same for all STM32 devices as far as I know, so this would make sence :)

lebrush, 2016-04-01:
The only issue I can think of is what @haukepetersen mentinoned in https://github.com/RIOT-OS/RIOT/pull/4926#issuecomment-200770061: every MCU might have different errata. 

One possible solution would be to have a main driver `i2c.c` which includes a "header" file `i2c_arch.c` which would be in the specific mcu folder:

main `i2c.c`
```C
...
#include "cpu.h"
#include "periph_conf.h"
#include "periph/i2c.h"
...
#if STM32_HAS_I2C_ARCH
#include "i2c_arch.c"
#endif
...
#ifndef STM32_I2C_ARCH_HAS_WRITE
int i2c_write_bytes(i2c_t dev, uint8_t address, char *data, int length) {
    ...
}
#endif
....
```

and in `i2c_arch.c`:
```C
#define STM32_I2C_ARCH_HAS_WRITE
int i2c_write_bytes(i2c_t dev, uint8_t address, char *data, int length) {
    ...
}
```

I just checked the errata sheets for L1 and F1 and they share some I2C issues but have also individual problems. The best would be to have a general overview of the possible erratas and see if supporting errors would cause problems in platforms which are not affected by them (which might be a bit dirty). Maybe the erratas are F series and L series dependent...

@DipSwitch @haukepetersen any thoughts?

haukepetersen, 2016-04-01:
In general unifying makes much sense, though as stated before: those minor changes in different register names and some different bit positions between the STM CPUs and on top of that different errata sheets don't make it necessarily nice code... Just take the DAC implementation as example: the implementation is normally dead simple, but these minor differences make the driver kind of ugly already with all those ifdefs...

So for the I2C driver I would tend to be very careful when joining them. If it leads to massively ifdefd code, nobody in the future will be able to maintain it and we would not win anything. But if we find a solution somewhere in the middle, I am all for it.

So I think by starting a diff-analysis for the I2C (and maybe also other periphs) is a good starting point here.

DipSwitch, 2016-04-05:
Another solution could be to ditch the header defines for STM32 and create our own defines for the registers.

And the DAC interface is not that ugly imo... But with more complicated peripherals (like i2c, adc etc) things could indeed get out of hand.

DipSwitch, 2016-04-05:
BTW I came across https://github.com/libopencm3/libopencm3/ maybe there we can get some ideas :)

gebart, 2016-04-06:
@DipSwitch I don't think it is a worthwhile investment of time to write register mapping header files for devices which we already have working maps. Making a custom memory map for the stm32 family will be a long term maintenance cost as support for new CPUs are added to RIOT.

haukepetersen, 2016-04-06:
Completely agree with @gebart here: writing the headers ourselves is not worth the effort (though we had to do it e.g. for the cc2538, as no CMSIS style headers are provided there...). Also the differences for STM CPUs are not only different naming schemes for register (which we can easily fix), but also minor annoying differences in register positions etc...

Regarding libopencm3: we had lenghtly discussions about using this ~2 years ago and came to the conclusion, that it does not improve our situation...

DipSwitch, 2016-04-19:
Ah ok, I missed that converstation hehe but makes sense :)

kYc0o, 2016-07-26:
Let's continue this discussion for the next release.

aabadie, 2018-10-19:
I just found this old one. The I2C rework brought an unify implementation of I2C driver for stm32. Since STM32 has 2 kinds of drivers for I2C, the strategy was to provide 2 different implementations based on the cpu type (F0/1/2/3/4/7, L0/1/4). This is similar to what was proposed by @lebrush.

I'm closing this one now but feel free to reopen if you disagree.

closed 2018-10-19 by aabadie