RIOT-OS/RIOT #5172: make clean is deleting package code

authmillenon, 2016-03-24:
It used to be `distclean` that did that, while `clean` just removed binaries and other files not in any commit.

kaspar030, 2016-03-24:
Why is that a bug?

authmillenon, 2016-03-24:
Because it requires an unnecessary download/copy (with git-cache) action. `distclean` for forcing re-cloning should suffice.

authmillenon, 2016-03-24:
PR to out it from an engineer's perspective: because it is an unexpected behavior from a previous state that was not solicited => a bug.

kaspar030, 2016-03-24:
> Because it requires an unnecessary download/copy (with git-cache) action. distclean for forcing re-cloning should suffice.

No way around that, as our normal "make clean" cleans the bin/$BOARD directory, unless we keep the pkg sources in a seperate directory. (bin/pkg/$board)? Will look into it.

kaspar030, 2016-03-24:
(with git-cache, the increased complexity is not worth the hassle...)

kaspar030, 2016-03-24:
> Or to out it from an engineer's perspective: because it is an unexpected behavior compared to a previous state that was not solicited => a bug.

Debatable, if the previous behaviour was, e.g., desastrous.

kaspar030, 2016-03-24:
I would prefer maybe automatic git-caching (without the need for manual setup/update). That way, all the ephemeral stuff stays in one board directory.

authmillenon, 2016-03-25:
Since when and why is a package cloned into an application's bindir?

kaspar030, 2016-03-25:
Since people and machines are trying to build multiple boards in parallel, in the same RIOTBASE.

PR #5174 (2016-03-25): pkg: (git): move pkgdir to $BINDIRBASE/pkg/$BOARD/$PKG_NAME, do proper clean
Fixes #5172 for git packages.
closed 2016-03-25 by authmillenon