RIOT-OS/RIOT #5183: CI: disconnect Travis

authmillenon, 2016-03-26:
Now that Murdock is our main CI and way better than Travis: Do we want to disconnect from Travis?

OlegHahm, 2016-03-27:
I would say: let's wait until after the release.

haukepetersen, 2016-03-29:
I am with Oleg here -> let's keep Travis attached for now, it does not hurt and we have a live fallback in case anything unexpected happens to Murdock...

cgundogan, 2016-03-29:
>I am with Oleg here -> let's keep Travis attached for now, it does not hurt and we have a live fallback in case anything unexpected happens to Murdock...

+1

authmillenon, 2016-03-29:
Though I don't want to invalidate your arguments I still would vote to disconnect Travis at some point in the future, to make successful builds again easy to find with https://github.com/RIOT-OS/RIOT/pulls?utf8=%E2%9C%93&q=is%3Apr+is%3Aopen+status%3Asuccess (when travis is producing noise)

authmillenon, 2016-05-17:
It's after the release and Travis seems only to fail now. Shall we disconnect it?

OlegHahm, 2016-05-17:
Fine with me.

authmillenon, 2016-05-17:
Done.

closed 2016-05-17 by authmillenon