RIOT-OS/RIOT #5197: Feature request: 6lowpan GHC support (RFC7400)

gebart, 2016-03-30:
Currently not implemented in the gnrc 6lowpan stack is the generic header compression (GHC) scheme as described in RFC 7400
http://tools.ietf.org/html/rfc7400

This is just a suggestion for a project in case anyone is looking for something to dig into.

OlegHahm, 2016-03-30:
AFAIK @ReneHerthel wanted to work on this. I have a non-RIOT implementation of GHC in https://github.com/OlegHahm/ghc, but it's not based on the RFC, but on an older version of the draft and it's untested (not even my own code).

kaspar030, 2016-03-30:
> I have a non-RIOT implementation of GHC in
> https://github.com/OlegHahm/ghc, but it's not based on the RFC, but on
> an older version of the draft and it's untested (not even my own code).

Sounds like fun! ;)

OlegHahm, 2016-03-30:
Well, to be honest some students of @emmanuelsearch hacked this long time ago and just wanted to make sure that the code isn't lost, but I cannot even tell if it makes sense to take this as a basis.

PeterKietzmann, 2016-03-30:
Nope, @ReneHerthel decided against implemention GHC but @jfischer-phytec-iot planned to implement it. @ReneHerthel is working on the implementation of elided checksums (part of NHC) in RIOT.

OlegHahm, 2016-03-30:
Thanks for the clarification, @PeterKietzmann. Sorry for my bad memory.

PeterKietzmann, 2016-03-30:
No worries. IIRC I didn't inform you about that decision :-).