RIOT-OS/RIOT #5201: RFC: use vagrant to test building on multiple systems

authmillenon, 2016-03-30:
I know, Vagrant was originally introduced to help users setting up a system to build RIOT in an easy way. But how about using it also to test building RIOT on multiple systems (different versions of Ubuntu and Debian, Arch, FreeBSD, and even MacOSX if we find a Vagrant image for that).

BytesGalore, 2016-03-30:
This sounds great in my ears, we would have direct proof if certain build environments have difficulties :+1:

kaspar030, 2016-03-30:
For the CI, we should build a limited set, like only unittests and maybe gnrc_networking, for native and a chosen board (or one board for each toolchain).

kYc0o, 2016-07-26:
Nice idea, but maybe we can realise it for the next release.

miri64, 2020-06-30:
Seems related https://github.com/RIOT-OS/RIOT/issues/3392

crasbe, 2026-10-07:
Vagrant has been removed in https://github.com/RIOT-OS/RIOT/pull/22711

closed 2026-10-07 by crasbe