RIOT-OS/RIOT #5207: make: buildest fails with board dependent application Makefiles

OlegHahm, 2016-03-30:
#3551 revealed that setting `$BOARD` dependent variables in the application Makefile breaks targets in `Makefile.buildtest` since the application Makefile gets executed without `$BOARD` being set.

Kijewski, 2016-03-30:
E.g. if `examples/some_app/Makefile` contains an `ifeq (${BOARD}, msba2)` section?

OlegHahm, 2016-03-30:
Exactly.

OlegHahm, 2016-03-30:
(Or maybe it's a little bit more complicated: if `examples/some_app/Makefile` contains an `ifeq (${BOARD}, msba2)` section with something like `FEATURE_REQUIRED` (or something else that should be evaluated by the Make system)).

haukepetersen, 2016-04-20:
this won't be fixed for the release -> postponed to the next one

kaspar030, 2016-07-11:
this won't be fixed for the release -> postponed to the next one

aabadie, 2017-07-07:
I tested this with openthread which has board specific makefile code and it works:
```
 $ make -C tests/openthread buildtest
Building for fox .. success
Building for iotlab-a8-m3 .. success
Building for iotlab-m3 .. success
Building for samr21-xpro .. success
make: Leaving directory '...RIOT/tests/openthread'
```

Is it related to what is described here ? If yes, I guess this one is fixed and we can close ?

cladmi, 2017-09-27:
I think this has been fixed by https://github.com/RIOT-OS/RIOT/pull/7589

miri64, 2017-10-25:
Yes, I also did not encounter this during release testing, so let's close this one.

closed 2017-10-25 by miri64