RIOT-OS/RIOT #5209: Board z1: Fix problem with UART port

Joelfp, 2016-03-30:
After recieving the new changes from #5072 I have problems with the UART port. While flashing the "hello world" program I recieve nothing from the UART. After viewing the UART port assign values in the file "RIOT/boards/z1/include/periph_conf.h" I noticed that `UART_RX_PORT` , `UART_TX_PORT` , `UART_TX_PIN` and `UART_RX_PIN` had assigned the incorrect values (lines 62-65). According to Zolertia z1 datasheet the UART TX is assigned to pin P3.4 and UART RX to pin P3.5.
After correcting these values I recieve the hello world message.

OlegHahm, 2016-03-30:
Do you care to open a PR for that?

Joelfp, 2016-03-30:
OK I will open a PR

OlegHahm, 2016-03-30:
Great!

haukepetersen, 2016-03-31:
did #5210 solve this issue? It seems like it can be closed, right?

Joelfp, 2016-03-31:
Yes, It solved the problem for me. I will close this issue

closed 2016-03-31 by Joelfp