RIOT-OS/RIOT #5223: boards/pba-d-01-kw2x: UART broken

haukepetersen, 2016-04-01:
Just wanted to run the default example on the `pba-d-01-kw2x` board - it seems broken, I don't get any response on the UART, not even the initial RIOT output. Same results for `examples/hello-world` and `tests/shell`...

cgundogan, 2016-04-01:
I just tested on current master. The shell works for me with the `default`, `hello-world` and `gnrc_networking` example. I also don't get RIOT's greeting right after `make term`, but after a reset it appears to work.

cgundogan, 2016-04-01:
For the sake of completeness, I tested with revision `1426.1`

haukepetersen, 2016-04-01:
now it works here as well, no idea what I did wrong earlier... So this issue does not exist -> closing it again.

closed 2016-04-01 by haukepetersen