RIOT-OS/RIOT #5230: gnrc ipv6: multicast packets are not dispatched to the upper layers

OlegHahm, 2016-04-01:
If I understand the forwarding in gnrc_ipv6 correctly multicast packets with a broader scope than link-local (e.g. global ones) are forwarded correctly, but are not passed to any upper layer. I would assume that upon reception of a multicast packet targeted for a global multicast address, the packet would be dispatched to the upper layers **and** forwarded.

authmillenon, 2016-04-01:
Huh, isn't `gnrc_ipc6_demux()` called for the multicast case?

OlegHahm, 2016-04-01:
If I see it correctly the function returns after the forwarding.

authmillenon, 2016-04-01:
Please rephrase...

OlegHahm, 2016-04-01:
The `_receive()` function of `gnrc_ipv6.c` returns after forwarding a packet in https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L912. Hence the demux call below will never be reached for this case.

authmillenon, 2016-04-01:
Ah sorry, I thought https://github.com/RIOT-OS/RIOT/pull/4528 (which fixes that) was already merged...

haukepetersen, 2016-04-20:
the fix (#4528) is set for 2016.07, so I do the same with this issue.

kYc0o, 2016-07-11:
Do you know if this issue is still present?

kYc0o, 2016-07-26:
Postponing again since the fix was postponed too.

PeterKietzmann, 2017-01-31:
This is the current PR that should fix it #5729

kYc0o, 2018-07-17:
According to #5729 this might have been solved in #7925. @miri64 do you confirm?

miri64, 2018-07-17:
Not sure. Remind me to test this.

miri64, 2018-07-17:
Mh... It seems like the reverse is the case now. Non-link-local multicast addresses are handed up the stack, but are not forwarded anymore :-/

miri64, 2018-07-17:
This is also kind of difficult to fix at the moment. I'd rather do this after I refactored IPv6 reception when #8511 is through.

maribu, 2023-05-24:
> This is also kind of difficult to fix at the moment. I'd rather do this after I refactored IPv6 reception when #8511 is through.

#8511 is through now :wink: