RIOT-OS/RIOT #5237: xtimer with any value clock frequency

attdona, 2016-04-03:
The xtimer module needs some patching for the (rare) case where frequency of the timer peripheral are not a natural one (1 usec) or a power of two of this frequency.

The real case is for example the fixed 80 Mhz frequency of cc3200 [1] timer module.

Below what works for this chip, at least with my preliminary tests, hoping this may help to find a rigorous patch that works for every system.

## MACRO for scaling ##
Redefined and moved the macros XTIMER_USEC_TO_TICKS and XTIMER_TICKS_TO_USEC 
into a specific header (boards/cc3200-launchxl/include/periph_conf.h:

    #define XTIMER_SHIFT 1
    #define XTIMER_USEC_TO_TICKS_FACTOR 80

    #define XTIMER_USEC_TO_TICKS(value) ( value*XTIMER_USEC_TO_TICKS_FACTOR )
    #define XTIMER_TICKS_TO_USEC(value) ( value/XTIMER_USEC_TO_TICKS_FACTOR )

## Syncronize wrapping of natural and hw counter ##
Aligned the wrap of natural counter (1 usec based) with the wrapping of the system timer (1/80 usec based)

    #ifdef XTIMER_SHIFT
      timer->target = (target - XTIMER_OVERHEAD)%(0xffffffff/XTIMER_USEC_TO_TICKS_FACTOR);
    #else
      timer->target = target - XTIMER_OVERHEAD;  //!!! to improve timer precision
     //timer->target = target;  //!!! this is RIOT original
    #endif

The only place where I did this is into xtimer_core.c/_xtimer_set_absolute function. I think some reviewing by the timer task force has to be done to confirm that this is correct.

As another minor consideration, not related to this issue, please see the //!!! comments for a suggested patch to improve the precision of timers as it comes out for my measures.

[1] http://www.ti.com/product/CC3200

OlegHahm, 2016-04-03:
Have you checked the ideas in #3990 and #4002?

attdona, 2016-04-04:
Just checked #3990 and it fits.

I don't know the maturity status of this PR, it is a major "rewriting": when ready I can test it against cc3200 board.

gebart, 2016-04-04:
The problem I have noticed with #3990 is that in its current state there are too many conversions because of the usec scaling of all API calls. I would like to create a stricter form of #4002 which checks for the correct xtimer tick types instead of a macro (which won't do type checking) and use internal ticks in all arguments and return values, to avoid having to convert back and forth in for example in the last_wakeup argument to `xtimer_usleep_until()`.

kYc0o, 2016-07-26:
This issue has been addressed in some PRs, but none of the has been merged or could be merged for this release. Let's try for the next one.

kYc0o, 2017-01-27:
"Any" it's a bit too much, but #5608 addressed this and #6507 takes care of 14.7456MHz frequencies.

closed 2019-09-10 by stale[bot]